johnfanv2/LenovoLegionLinux #252: Installation failed lenovolegionlinux-dkms-git

0xSushmanth, 2024-09-08:
## Issue: Installation Failure on Lenovo Legion 15ACH6

### Description

I'm encountering an issue when trying to install the `lenovolegionlinux-dkms-git` package from the [AUR](https://aur.archlinux.org/packages/lenovolegionlinux-dkms-git) on my Lenovo Legion 15ACH6. The installation fails with the following error messages:

`ERROR: A failure occurred in build(). Aborting... error: failed to build 'lenovolegionlinux-git-r255.680ed61-1': error: can't build lenovolegionlinux-dkms-git-r255.ef427ef-0, deps not satisfied: lenovolegionlinux-git error: packages failed to build: lenovolegionlinux-git-r255.680ed61-1 lenovolegionlinux-dkms-git-r255.ef427ef-0`

It appears that the `lenovolegionlinux-git` dependency is not being satisfied, which is preventing the successful build of the `lenovolegionlinux-dkms-git` package.

### System Information

- **Laptop Model:** Lenovo Legion 15ACH6
- **OS:** CachyOS
- **Host:** 82JW (NB LN Legion 5 15ACH6)
- **Kernel:** Linux 6.10.8-2-cachyos

johnfanv2, 2024-09-08:
Maybe @MrDuartePT can shed some light onto it. Does it output any more errors, e.g. errors from compiling the kernel module. There was some change in the kernel API in the new Linux kernel that caused that some problems which I just fixed on branch `main`.

MrDuartePT, 2024-09-08:
That is happen because you are installing the aur package manually and didn't install `LenovoLegionLinux-git` frist, that you are having missing dependency error.

If you don't know how to manually manage dependencies of the aur, just use a helper like yay or paru.

closed 2024-09-08 by MrDuartePT