johnfanv2/LenovoLegionLinux #262: Legion 5 Slim Gen 9 Support?

TechLionDev, 2024-09-24:
**Is your feature request related to a problem? Please describe.**
No, was just wondering if the new Legion 5 Slim Gen 9 was supported

**Describe the solution you'd like**
to be supported
**Describe alternatives you've considered**
None ATM

**NOTE: IDK because its in shipping i just wanna be ready since ik im gonna use linux since im a developer**

**Model and Debug Info**

Distribution: CachyOS (Arch)
Model name: Legion Slim 5 Gen 9 AMD (16″) with RTX 4070
CPU model: AMD Ryzen™ 7 8845HS Processor
GPU model: NVIDIA® GeForce RTX™ 4070 Laptop GPU 8GB GDDR6
Keyboard backlight: 4zone RGB Backlit, Black - English (US)
Light in lid or logo: IDK
Light at IO-Ports at back: IDK

Output of `sudo dmidecode -t system`. Please remove Serial Number and UUID for privacy:
```text
IDK
```

Output of `sudo dmidecode -t bios`:
```text
IDK
```

Output of `sudo cat /sys/kernel/debug/legion/fancurve`:
```text
IDK
```

TechLionDev, 2024-09-24:
> Hey that's my computer right there! No it's not supported, see #218
> Unfortunately there hasn't been a new supported device for quite some now,
> last one being in january

Are there any other projects that offer what lenovo vantage does, but for
linux, and if I configure in windows before I install Linux will those
settings persist?

Evthestrike, 2024-10-07:
I would love this!

closed 2026-08-28 by TechLionDev