johnfanv2/LenovoLegionLinux #265: Legion Slim 5 16APH8 - Type 82Y9 support

Lontor, 2024-10-05:
**Problem Description**

 1. Driver Loading Error
When running `sudo make reloadmodule`, I encounter the following error:
```
legion_laptop: Loading legion_laptop
legion PNP0C09:00: legion_laptop platform driver probing
legion PNP0C09:00: Read identifying information: DMI_SYS_VENDOR: LENOVO; DMI_PRODUCT_NAME: 82Y9; DMI_BIOS_VERSION:M3CN43WW
legion_laptop: Read identifying information: DMI_SYS_VENDOR: LENOVO; DMI_PRODUCT_NAME: 82Y9; DMI_BIOS_VERSION:M3CN43WW
legion PNP0C09:00: is_denied: 0; is_allowed: 1; do_load_by_list: 1; do_load: 1
legion PNP0C09:00: Using configuration for system: M3CN
legion PNP0C09:00: Could not evaluate ACPI _STA
legion PNP0C09:00: Could not init ACPI access: -5
legion_laptop: Unloading legion shared
legion_laptop: Unloading legion shared done
legion PNP0C09:00: legion_laptop not loaded for this device
legion PNP0C09:00: probe with driver legion failed with error -5
```
However, when I run `sudo make forcereloadmodule`, the driver loads without errors, but there are issues with certain features, as described below.
1. The **Fan Curve** menu is active, but when I try to read or apply settings, the application crashes with the following errors:
Error on "Read from HW":
```
Traceback (most recent call last):
  File "/usr/lib/python3.12/site-packages/legion_linux/legion_gui.py", line 869, in on_read_fan_curve_from_hw
    self.model.read_fancurve_from_hw()
  File "/usr/lib/python3.12/site-packages/legion_linux/legion.py", line 1469, in read_fancurve_from_hw
    self.fan_curve = self.fancurve_io.read_fan_curve()
                     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/site-packages/legion_linux/legion.py", line 942, in read_fan_curve
    fan1_speed = self.get_fan_1_speed(point_id)
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/site-packages/legion_linux/legion.py", line 848, in get_fan_1_speed
    return self._read_file(file_path)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/site-packages/legion_linux/legion.py", line 777, in _read_file
    return int(filepointer.read())
               ^^^^^^^^^^^^^^^^^^
OSError: [Errno 95] Operation not supported
```
Error on "Write to HW":
```
Traceback (most recent call last):
  File "/usr/lib/python3.12/site-packages/legion_linux/legion_gui.py", line 875, in on_write_fan_curve_to_hw
    self.model.read_fancurve_from_hw()
  File "/usr/lib/python3.12/site-packages/legion_linux/legion.py", line 1469, in read_fancurve_from_hw
    self.fan_curve = self.fancurve_io.read_fan_curve()
                     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/site-packages/legion_linux/legion.py", line 942, in read_fan_curve
    fan1_speed = self.get_fan_1_speed(point_id)
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/site-packages/legion_linux/legion.py", line 848, in get_fan_1_speed
    return self._read_file(file_path)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/site-packages/legion_linux/legion.py", line 777, in _read_file
    return int(filepointer.read())
               ^^^^^^^^^^^^^^^^^^
OSError: [Errno 95] Operation not supported
```

- **Fn Lock**: Working
- **Win Key Enabled**: Working
- **Touchpad Enabled**: Working
- **Camera Power Enabled**: Not active (this model has a physical switch for camera)
- **Battery Conservation**: Appears to be working
- **Rapid Charging**: Appears to be working
- **Charge Output from USB Always On**: Causes an error:
```
INFO:legion_linux.legion:Feature AlwaysOnUSBChargingFeature reading
INFO:legion_linux.legion:Feature AlwaysOnUSBChargingFeature reading: 1
INFO:legion_linux.legion:Click on UI Charge Output from U&SB always on element for AlwaysOnUSBChargingFeature
[... 169 lines cut ...]
has custom powermode: 1
ACPI rapidcharge error: 0
ACPI rapidcharge: 0
WMI backlight 2 state: 0
WMI backlight 3 state: 0
WMI light IO port: 1
WMI light Y logo/lid: 1
EC minifancurve feature enabled: 0
EC minifancurve on cool: error
EC lockfancontroller error: 0
EC lockfancontroller: false
fanfullspeed error: 0
fanfullspeed: 0
fanfullspeed EC error: 0
fanfullspeed EC: 0
EC fan curve current point id: 0
EC fan curve points size: 0
Current fan curve in hardware:
rpm1|rpm2|acceleration|deceleration|cpu_min_temp|cpu_max_temp|gpu_min_temp|gpu_max_temp|ic_min_temp|ic_max_temp
=====================
Current fan curve in hardware (WMI; might be empty)
rpm1|rpm2|acceleration|deceleration|cpu_min_temp|cpu_max_temp|gpu_min_temp|gpu_max_temp|ic_min_temp|ic_max_temp
=====================
```
### If you need any further information or logs, feel free to ask! Thank you for looking into this issue.

deebfeast, 2025-04-25:
Exactly the same here. FYI as a work-around, I obtained a reasonable stable Linux fan behavior by:

- using the latest BIOS
- disabling tlp service on kernel 6.8.0 as it interfered with hardware control
- Fn + q allows for manual switching between auto, quiet and performance
 
I'm also running in performance mode (prime-select nvidia) as amd-gpu was behaving badly but the fans still are behaving.

conqazht, 2026-08-28:
@Lontor  @deebfeast Has the issue been resolved yet?

deebfeast, 2026-08-28:
> [@Lontor](https://github.com/Lontor) [@deebfeast](https://github.com/deebfeast) Has the issue been resolved yet?

Just tried 0.0.21 instead of 0.0.18

```
sudo make reloadmodule

PNP0C09:00: Read embedded controller ID 0x8227
PNP0C09:00: Expected EC chip id 0x5507 but read 0x8227
legion_laptop: Unloading legion ecram
legion_laptop: Unloading legion ecram done
legion_laptop: Unmapping embedded controller memory at 0xfe0b0400 (in RAM)/0x0 (in EC) at virtual 0x000000003b627505
legion_laptop: Unloading legion shared
legion_laptop: Unloading legion shared done
PNP0C09:00: legion_laptop not loaded for this device
```

It does load with make reloadmoduleforced and I can read fan speed out but not change anything. I think the different EC chip causes a lot of things to not work, show up empty or not doing anything when trying to echo/enable anything. The table that shows the current fan curve is still mostly zero. Haven't looked a lot further. Also because I'm very happy with my fans managed by the hardware now. Although the thermal profile of such laptops is such that running CPU/GPU intensive software will be very noisy to keep the processor under throttling.

conqazht, 2026-08-28:
> Just tried 0.0.21 instead of 0.0.18
>
> ```
> sudo make reloadmodule
>
> PNP0C09:00: Read embedded controller ID 0x8227
> PNP0C09:00: Expected EC chip id 0x5507 but read 0x8227
> legion_laptop: Unloading legion [...]

How are you currently managing the fans? When my laptop's temperature rises, the fans don't work as well as they do on Windows, especially when using Linux; the machine always gets hotter than when using Windows.

deebfeast, 2026-08-28:
> > > [@Lontor](https://github.com/Lontor) [@deebfeast](https://github.com/deebfeast) Has the issue been resolved yet?
> How are you currently managing the fans? When my laptop's temperature rises, the fans don't work as well as they do on Windows, especially when using Linux; the machine always gets hotter than when using Windows.

I'm not managing anymore. The first laptop since many where I don't do anything special anymore for fans under Linux. 

Under Windows the Lenovo tools are likely loaded automatically. What profile do you use there? When using only hardware/BIOS, I use Fn + q for manual switching between auto, quiet and performance. For me _performance_ is the quietest strangely  enough (it shows orange led). The other two modes seem to change spin up the fan at some interval, not sure if it keeps actually things cooler.

My linux kernel now is 6.8.0-138-generic although it's been good for a year of updates. 

Make sure that demanding software is actually using nvidia or use e.g. "prime-select nvidia" to make sure it's always used. The amdgpu gets a bit hotter when running some stuff. Under Windows the auto-switching to nvidia seems often better integrated.

Laptop BIOS: Version: M3CN44WW Release Date: 10/07/2024. There's a newer out even but only for security updates. I do think the update to this version improved quite a bit the cooling through hardware.

st0nie, 2026-09-14:
The Slim 5 16APH8 / 82Y9 is covered: the product-qualified entry "Legion R7000P APH8 (82Y9) - EC 0x8227" (M3CN BIOS, #503) matches your DMI data. Please retest with the current release and reopen with full dmesg if the load error persists.

closed 2026-09-14 by st0nie