johnfanv2/LenovoLegionLinux #273: [BUG] 0.0.18 Compilation Error in DKMS Module Due to Incompatibility with Python 3.13, DNF5, or Recent Kernel (most likely).

Allephh, 2024-11-02:
**Problem Description**
The LenovoLegionLinux driver was probably not installed, since nothing has changed in practice. Although the Python GUI is here, I can't change anything there, and the options are impossible to activate.
At first, I thought it was a problem related to Python, because in the pre-installation, the text said **Python(abi) = 3.12** was missing. Even installing this version of Python (along with pip = darkdetect) didn't solve the problem.
Before, in the post-installation, there were also problems related to DNF5, where it showed possible commands that had changed or no longer existed. However, when I installed it again yesterday after formatting, this didn't happen. Log of recently installation:
[sudo.dnf.install.dkms.txt](https://github.com/user-attachments/files/17608552/sudo.dnf.install.dkms.txt)

And that's where the Kernel comes in.
**When I start the system, it presents problems like these:**
systemd-backlightglade:pi_s (or se) of leds: platform: :kbd_backlight.
Failed to start: dkms.service - Bui...ll new kernel modules through DKMS. 'systemctl status dkms.service' for details.

**Here are the videos demonstrating the absence of the driver and the errors.**
https://github.com/user-attachments/assets/9081bd86-21d6-4442-bdd3-cdf122db0b5b
https://github.com/user-attachments/assets/23e0b24b-6b51-4c6c-bd0d-6041f48aa19a
https://github.com/user-attachments/assets/0615c66c-c945-40ab-a7e7-2df9e7b09b4d
https://github.com/user-attachments/assets/2d7123c3-cab4-4924-a943-54c6167b6372

**And here are the terminal commands that show the logs:**
[image]
[image]

**Hardware Information:**
- **Hardware Model:**                     Lenovo IdeaPad Gaming 3 15ARH7
- **Memory:**                                    16.0 GiB
- **Processor:**                                  AMD Ryzen™ 7 6800H with Radeon™ Graphics × 16
- **Graphics:**                                    AMD Radeon™ 680M
- **Graphics 1:**                                 NV177
- **Disk Capacity:**                            1 TB.

**Software Information:**
- **Firmware Version:**                     JNCN51WW(V2.11)
- **OS Name:**                                   Fedora Linux 41 (Workstation Edition)       
- **OS Type:**                                     64-bit
- **GNOME Version:**                       47
- **Windowing System:**                  Wayland
- **Kernel Version:**                          Linux 6.11.5-300.fc41.x86_64

johnfanv2, 2024-11-04:
That is using the old version. The compilation error for the "remove" function was fixed on master. A new version of the pkg for Fedora is needed.

Allephh, 2024-11-05:
> That is using the old version. The compilation error for the "remove" function was fixed on master. A new version of the pkg for Fedora is needed.

Hi, sorry for the late.
What you mean by this? Sorry for my ignorance

RodrigoPrestes, 2024-11-17:
> That is using the old version. The compilation error for the "remove" function was fixed on master. A new version of the pkg for Fedora is needed.

Any ideas when we plan to tag a new version?

> Hi. Could it have anything to do with this? [Changes/TunedAsTheDefaultPowerProfileManagementDaemon](https://fedoraproject.org/wiki/Changes/TunedAsTheDefaultPowerProfileManagementDaemon)

If that's the case, that is being tracked here I believe: https://github.com/johnfanv2/LenovoLegionLinux/issues/279

RodrigoPrestes, 2024-11-19:
Looks like we have a [new tag](https://github.com/johnfanv2/LenovoLegionLinux/releases/tag/v0.0.19), waiting new version to show up on copr.

MrDuartePT, 2024-11-23:
@Allephh I build new packages just now pls test:
[image]

RodrigoPrestes, 2024-11-23:
Tested on a Legion Slim 7 (16ARHA7, AMD Advantage) on a clean install.
v0.0.19 seems to have fixed installation on Fedora Workstation 41 as dnf completed without errors, and we closed https://github.com/johnfanv2/LenovoLegionLinux/issues/283.

--- 

On the other hand I couldn't make it work, it even breaks things, after a reboot it was using software rendering.
I've been trying to install these past few days on Fedora Workstation 40 with v0.0.18 and had the same issue so maybe I just don't know how to set it up on Fedora.

Here's the log where it has errors, although that's just the GUI:
```
nov 23 09:01:00 fedora legion_gui.desktop[3469]: Error occured cannot access local variable 'values' where it is not associated with a value
nov 23 09:01:00 fedora legion_gui.desktop[3469]: Traceback (most recent call last):
nov 23 09:01:00 fedora legion_gui.desktop[3469]:   File "/usr/lib/python3.13/site-packages/legion_linux/legion_gui.py", line 397, in update_view_from_feature
nov 23 09:01:00 fedora legion_gui.desktop[3469]:     if i< len(values):
nov 23 09:01:00 fedora legion_gui.desktop[3469]:               ^^^^^^
nov 23 09:01:00 fedora legion_gui.desktop[3469]: UnboundLocalError: cannot access local variable 'values' where it is not associated with a value
nov 23 09:01:00 fedora legion_gui.desktop[3469]: Error occured cannot access local variable 'values' where it is not associated with a value
nov 23 09:01:00 fedora legion_gui.desktop[3469]: Traceback (most recent call last):
nov 23 09:01:00 fedora legion_gui.desktop[3469]:   File "/usr/lib/python3.13/site-packages/legion_linux/legion_gui.py", line 397, in update_view_from_feature
nov 23 09:01:00 fedora legion_gui.desktop[3469]:     if i< len(values):
nov 23 09:01:00 fedora legion_gui.desktop[3469]:               ^^^^^^
nov 23 09:01:00 fedora legion_gui.desktop[3469]: UnboundLocalError: cannot access local variable 'values' where it is not associated with a value
nov 23 09:01:00 fedora systemd[2468]: app-gnome-legion_gui-3469.scope: Consumed 719ms CPU time, 74.8M memory peak.
nov 23 09:16:19 fedora kernel: legion_laptop: loading out-of-tree module taints kernel.
nov 23 09:16:19 fedora kernel: legion_laptop: module verification failed: signature and/or required key missing - tainting kernel
```
Maybe we can close this and I'll open a new issue since it seems to be a different problem.

MrDuartePT, 2024-11-23:
That error is very strange and dosent happen on my system

closed 2024-11-23 by MrDuartePT

Allephh, 2024-11-23:
> @Allephh I build new packages just now pls test: [...]

[image]
Hello!
Should i proceed?

RodrigoPrestes, 2024-11-23:
@Allephh personally I didn't explicitly installed `python-darkdetect` since it's a dependency.
Can you try just `sudo dnf install dkms-LenovoLegionLinux python-LenovoLegionLinux`? (even though it's bound to fail as it's going to try to install `python-darkdetect` anyways)

Since I just tested with a clean install and it did not complained like this, maybe check if you have anything python related installed that might be causing this error.

Allephh, 2024-11-23:
> @Allephh personally I didn't explicitly installed `python-darkdetect` since it's a dependency. Can you try just `sudo dnf install dkms-LenovoLegionLinux python-LenovoLegionLinux`? (even though it's [...]

Apparently everything is fine. Fn + Q working. 
[image]

Allephh, 2024-11-23:
Log of information that the LenovoLegionLinuxGui application reported in the corresponding section. 
I think now i'll just disable the icon, install the new NVIDIA driver and start playing.
Thanks everyone.

2024-11-23 20:17:33,126 - INFO: 
2024-11-23 20:17:33,126 - INFO: Feature LockFanController with pattern /sys/module/legion_laptop/drivers/platform:legion/PNP0C09:00/lockfancontroller and path /sys/module/legion_laptop/drivers/platform:legion/PNP0C09:00/lockfancontroller
2024-11-23 20:17:33,126 - INFO: Feature RapidChargingFeature with pattern /sys/module/legion_laptop/drivers/platform:legion/PNP0C09:00/rapidcharge and path /sys/module/legion_laptop/drivers/platform:legion/PNP0C09:00/rapidcharge
2024-11-23 20:17:33,126 - INFO: Feature BatteryConservation with pattern /sys/bus/platform/drivers/ideapad_acpi/VPC2004:00/conservation_mode and path /sys/bus/platform/drivers/ideapad_acpi/VPC2004:00/conservation_mode
2024-11-23 20:17:33,127 - INFO: Feature MaximumFanSpeedFeature with pattern /sys/module/legion_laptop/drivers/platform:legion/PNP0C09:00/fan_fullspeed and path /sys/module/legion_laptop/drivers/platform:legion/PNP0C09:00/fan_fullspeed
2024-11-23 20:17:33,127 - INFO: Feature FnLockFeature with pattern /sys/bus/platform/drivers/ideapad_acpi/VPC2004:00/fn_lock and path /sys/bus/platform/drivers/ideapad_acpi/VPC2004:00/fn_lock
2024-11-23 20:17:33,127 - INFO: Feature WinkeyFeature with pattern /sys/module/legion_laptop/drivers/platform:legion/PNP0C09:00/winkey and path /sys/module/legion_laptop/drivers/platform:legion/PNP0C09:00/winkey
2024-11-23 20:17:33,127 - INFO: Feature TouchpadFeature with pattern ['/sys/bus/platform/drivers/ideapad_acpi/VPC2004:00/touchpad', '/sys/module/legion_laptop/drivers/platform:legion/PNP0C09:00/touchpad'] and path /sys/module/legion_laptop/drivers/platform:legion/PNP0C09:00/touchpad
2024-11-23 20:17:33,127 - INFO: Feature CameraPowerFeature with pattern /sys/bus/platform/drivers/ideapad_acpi/VPC2004:00/camera_power and path None
2024-11-23 20:17:33,127 - WARNING: Feature CameraPowerFeature does not exist. exits: 0
2024-11-23 20:17:33,127 - INFO: Feature OverdriveFeature with pattern /sys/module/legion_laptop/drivers/platform:legion/PNP0C09:00/overdrive and path /sys/module/legion_laptop/drivers/platform:legion/PNP0C09:00/overdrive
2024-11-23 20:17:33,127 - INFO: Feature GsyncFeature with pattern /sys/module/legion_laptop/drivers/platform:legion/PNP0C09:00/gsync and path /sys/module/legion_laptop/drivers/platform:legion/PNP0C09:00/gsync
2024-11-23 20:17:33,127 - INFO: Feature PlatformProfileFeature with pattern /sys/firmware/acpi/platform_profile and path /sys/firmware/acpi/platform_profile
2024-11-23 20:17:33,127 - INFO: Feature StrFileFeature with pattern /sys/firmware/acpi/platform_profile_choices and path /sys/firmware/acpi/platform_profile_choices
2024-11-23 20:17:33,127 - INFO: Feature IsOnPowerSupplyFeature with pattern /sys/class/power_supply/ADP0/online and path None
2024-11-23 20:17:33,127 - WARNING: Feature IsOnPowerSupplyFeature does not exist. exits: 0
2024-11-23 20:17:33,128 - INFO: Feature AlwaysOnUSBChargingFeature with pattern /sys/bus/platform/drivers/ideapad_acpi/VPC2004:00/usb_charging and path None
2024-11-23 20:17:33,128 - WARNING: Feature AlwaysOnUSBChargingFeature does not exist. exits: 0
2024-11-23 20:17:33,128 - INFO: Feature BatteryCurrentCapacityPercentage with pattern /sys/class/power_supply/BAT0/capacity and path None
2024-11-23 20:17:33,128 - WARNING: Feature BatteryCurrentCapacityPercentage does not exist. exits: 0
2024-11-23 20:17:33,128 - INFO: Feature CPUOverclock with pattern /sys/module/legion_laptop/drivers/platform:legion/PNP0C09:00/cpu_oc and path /sys/module/legion_laptop/drivers/platform:legion/PNP0C09:00/cpu_oc
2024-11-23 20:17:33,128 - INFO: Feature CPULongtermPowerLimit with pattern /sys/module/legion_laptop/drivers/platform:legion/PNP0C09:00/cpu_longterm_powerlimit and path /sys/module/legion_laptop/drivers/platform:legion/PNP0C09:00/cpu_longterm_powerlimit
2024-11-23 20:17:33,128 - INFO: Feature CPUShorttermPowerLimit with pattern /sys/module/legion_laptop/drivers/platform:legion/PNP0C09:00/cpu_shortterm_powerlimit and path /sys/module/legion_laptop/drivers/platform:legion/PNP0C09:00/cpu_shortterm_powerlimit
2024-11-23 20:17:33,128 - INFO: Feature CPUPeakPowerLimit with pattern /sys/module/legion_laptop/drivers/platform:legion/PNP0C09:00/cpu_peak_powerlimit and path /sys/module/legion_laptop/drivers/platform:legion/PNP0C09:00/cpu_peak_powerlimit
2024-11-23 20:17:33,128 - INFO: Feature CPUDefaultPowerLimit with pattern /sys/module/legion_laptop/drivers/platform:legion/PNP0C09:00/cpu_default_powerlimit and path /sys/module/legion_laptop/drivers/platform:legion/PNP0C09:00/cpu_default_powerlimit
2024-11-23 20:17:33,128 - INFO: Feature CPUCrossLoadingPowerLimit with pattern /sys/module/legion_laptop/drivers/platform:legion/PNP0C09:00/cpu_cross_loading_powerlimit and path /sys/module/legion_laptop/drivers/platform:legion/PNP0C09:00/cpu_cross_loading_powerlimit
2024-11-23 20:17:33,128 - INFO: Feature CPUAPUSPPTPowerLimit with pattern /sys/module/legion_laptop/drivers/platform:legion/PNP0C09:00/cpu_apu_sppt_powerlimit and path /sys/module/legion_laptop/drivers/platform:legion/PNP0C09:00/cpu_apu_sppt_powerlimit
2024-11-23 20:17:33,128 - INFO: Feature GPUOverclock with pattern /sys/module/legion_laptop/drivers/platform:legion/PNP0C09:00/gpu_oc and path /sys/module/legion_laptop/drivers/platform:legion/PNP0C09:00/gpu_oc
[... cut ...]
2024-11-23 20:17:33,477 - INFO: Feature PlatformProfileFeature reading
2024-11-23 20:17:33,540 - INFO: Feature PlatformProfileFeature reading: quiet
2024-11-23 20:17:33,541 - INFO: update_view_from_feature in EnumFeatureTrayController 
2024-11-23 20:17:33,542 - INFO: Feature StrFileFeature reading
2024-11-23 20:17:33,542 - INFO: Feature StrFileFeature reading: quiet balanced performance
2024-11-23 20:17:33,542 - INFO: Feature PlatformProfileFeature reading
2024-11-23 20:17:33,606 - INFO: Feature PlatformProfileFeature reading: quiet

Allephh, 2024-11-23:
I think now i'll just disable the icon, install the new NVIDIA driver and start playing.
Thanks everyone.

Log of information that the LenovoLegionLinuxGui application reported in the corresponding section. 
[2024 11 23 20 17 33 126.txt](https://github.com/user-attachments/files/17883093/2024.11.23.20.17.33.126.txt)

Allephh, 2024-11-24:
After playing a bit, i have something to report.
I don't know if it's the LenovoLegionLinux driver's fault, but my NVIDIA video card seems to be super limited.
The fans work, but the temperatures - which should be high - don't, no matter the power mode. I don't know if it's something related to Fedora 41 and its native Wayland, but the performance in games is not as high as in Fedora 40. It seems like i'm using none or very little of it, in other means, only the APU. However, the system seems to identify it (icon, system details).
If i prove anything, i'll open a new discussion.

Allephh, 2024-11-25:
I haven't been able to prove it concretely yet. It's something related to Wayland or Tuned.
As for Tuned, there are many Fedora 41 users complaining about poor performance and correlating it with that. Maybe it's a good idea to take a look.
When i turned on my computer a little while ago, i pressed ALT + TAB and some very strange errors appeared. The line was yellow and when GDM loaded, it was incomplete. I restarted and everything worked.

[image]

MrDuartePT, 2024-11-25:
That don't have nothing to do with our driver. That is a problem with you fedora installation