johnfanv2/LenovoLegionLinux #279: [BUG] LLL uses power-profiles-daemon but Fedora upstream has replaced it with tuned-ppd

GloriousEggroll, 2024-11-13:
https://fedoraproject.org/wiki/Changes/TunedAsTheDefaultPowerProfileManagementDaemon
This happened in 41

johnfanv2, 2024-11-14:
@MrDuartePT Do you know any details about that?

We still expose the profile from ACPI here /sys/firmware/acpi/platform_profile? Shouldn't tuned also pick it up. The only difference is in the GUI, python, and deamon that might call the power profile manager.

MrDuartePT, 2024-11-14:
> @MrDuartePT Do you know any details about that?
> 
Not really I don't use fedora in my personal machine, need to give a look 
> 
> We still expose the profile from ACPI here /sys/firmware/acpi/platform_profile? Shouldn't tuned also pick it up. The only difference is in the GUI, python, and deamon that might call the power profile manager.

For what i know `/sys/firmware/acpi/platform_profile` is the default location for platform_profile I use tlp on my machine and it can adjust the legion power profile.

The daemon it was updated to not required PPD if I not wrong.

MrDuartePT, 2024-11-14:
Ok I was correct the daemon was update to not required ppd: https://github.com/johnfanv2/LenovoLegionLinux/commit/ef427eff9b55f62ce7d23165bd49f6811c92ad78

@johnfanv2 what we could do is removed the toogle to enable ppd in the GUI or make it check if the system have PPD, TLP or tuned-ppd install. 
I think all distro enable ppd by default anyway so the toogle might be not needed. We also should update to README say it that the driver works with any PowerProfile Management daemon.

closed 2025-05-11 by st0nie