johnfanv2/LenovoLegionLinux #280: [BUG] ac_path wrong on LOQ 15IAX9

random2907, 2024-11-15:
**Problem Description**
<Please describe your problem here> 
Has the bug occurred following steps described in the README? No
What steps did you take? 
I debug by coredump and find the ac_path points to ADP0 but in my system it should be pointing to ACAD in file (extra/service/legiond/modules/powerstate.h)
What commands did you enter? sudo legiond
What was the (full) error you received and what command or action did lead to it? 
---------------set_all start---------------
zsh: segmentation fault  sudo legiond

**Model and Debug Info**

Installation type: AUR lenovolegionlinux-dkms-git
Distribution: Archlinux
Model name: LOQ 15IAX9
CPU model: Intel Core i5-12450HX
GPU model: NVIDIA RTX 2050
Keyboard backlight: single color with off/medium/bright
Light in lid or logo: yes
Light at IO-Ports at back: yes

Output of `sudo dmidecode -t system`. Please remove Serial Number and UUID for privacy:
```text
# dmidecode 3.6
Getting SMBIOS data from sysfs.
SMBIOS 3.4 present.

Handle 0x0001, DMI type 1, 27 bytes
System Information
	Manufacturer: LENOVO
	Product Name: 83GS
	Version: LOQ 15IAX9
	Serial Number: MP2N678M
	UUID: 1885662c-d60b-11ee-a4f6-40c2ba3cba6e
	Wake-up Type: Power Switch
	SKU Number: LENOVO_MT_83GS_BU_idea_FM_LOQ 15IAX9
	Family: LOQ 15IAX9

Handle 0x0024, DMI type 12, 5 bytes
System Configuration Options
	Option 1: ConfigOptions1
	Option 2: ConfigOptions2
	Option 3: ConfigOptions3

Handle 0x0027, DMI type 15, 29 bytes
System Event Log
	Area Length: 0 bytes
	Header Start Offset: 0x0000
	Header Length: 8192 bytes
	Data Start Offset: 0x2000
	Access Method: General-purpose non-volatile data functions
	Access Address: 0x0000
	Status: Valid, Not Full
	Change Token: 0x12345678
	Header Format: OEM-specific
	Supported Log Type Descriptors: 3
	Descriptor 1: POST memory resize
	Data Format 1: None
	Descriptor 2: POST error
	Data Format 2: POST results bitmap
	Descriptor 3: Log area reset/cleared
	Data Format 3: None

Handle 0x0033, DMI type 32, 11 bytes
System Boot Information
	Status: No errors detected
```

Output of `sudo dmidecode -t bios`:
```text
# dmidecode 3.6
Getting SMBIOS data from sysfs.
SMBIOS 3.4 present.

Handle 0x0000, DMI type 0, 26 bytes
BIOS Information
	Vendor: LENOVO
	Version: NECN38WW
	Release Date: 07/19/2024
	Address: 0xE0000
	Runtime Size: 128 kB
	ROM Size: 16 MB
	Characteristics:
		PCI is supported
		BIOS is upgradeable
		BIOS shadowing is allowed
		Boot from CD is supported
		Selectable boot is supported
		EDD is supported
		Japanese floppy for NEC 9800 1.2 MB is supported (int 13h)
		Japanese floppy for Toshiba 1.2 MB is supported (int 13h)
		5.25"/360 kB floppy services are supported (int 13h)
		5.25"/1.2 MB floppy services are supported (int 13h)
		3.5"/720 kB floppy services are supported (int 13h)
		3.5"/2.88 MB floppy services are supported (int 13h)
		8042 keyboard services are supported (int 9h)
		CGA/mono video services are supported (int 10h)
		ACPI is supported
		USB legacy is supported
		BIOS boot specification is supported
		Targeted content distribution is supported
		UEFI is supported
	BIOS Revision: 1.38
	Firmware Revision: 1.38

Handle 0x0025, DMI type 13, 22 bytes
BIOS Language Information
	Language Description Format: Long
	Installable Languages: 8
		en|US|iso8859-1,0
		fr|FR|iso8859-1,0
		zh|TW|unicode,0
		ja|JP|unicode,0
		it|IT|iso8859-1,0
		es|ES|iso8859-1,0
		de|DE|iso8859-1,0
		pt|PT|iso8859-1,0
	Currently Installed Language: en|US|iso8859-1,0
```

Output of `sudo cat /sys/kernel/debug/legion/fancurve`:
```text
EC Chip ID: 8227
EC Chip Version: 2a4
legion_laptop features: fancurve powermode platformprofile platformprofilenotify minifancurve
legion_laptop ec_readonly: 0
ACPI CFG error: 0
ACPI CFG: 4245487636
temperature access method: 5
CPU temperature error: 0
CPU temperature: 41
CPU temperature EC error: 0
CPU temperature EC: 0
CPU temperature ACPI error: -5
CPU temperature ACPI: 0
CPU temperature WMI error: 0
CPU temperature WMI: 41
CPU temperature WMI2 error: 0
CPU temperature WMI2: 0
CPU temperature WMI3 error: 0
CPU temperature WMI3: 41
GPU temperature error: 0
GPU temperature: 35
GPU temperature EC error: 0
GPU temperature EC: 0
GPU temperature ACPI error: -5
GPU temperature ACPI: 0
GPU temperature WMI error: 0
GPU temperature WMI: 35
GPU temperature WMI2 error: 0
GPU temperature WMI2: 0
GPU temperature WMI3 error: 0
GPU temperature WMI3: 35
fan speed access method: 5
1 fanspeed error: 0
1 fanspeed: 1600
1 fanspeed EC error: 0
1 fanspeed EC: 41120
1 fanspeed ACPI error: -5
1 fanspeed ACPI: 41120
1 fanspeed WMI error: 0
1 fanspeed WMI: 0
1 fanspeed WMI2 error: 0
1 fanspeed WMI2: 0
1 fanspeed WMI3 error: 0
1 fanspeed WMI3: 1600
2 fanspeed error: 0
2 fanspeed: 1900
2 fanspeed EC error: 0
2 fanspeed EC: 41120
2 fanspeed ACPI error: 0
2 fanspeed ACPI: 1900
2 fanspeed WMI error: 0
2 fanspeed WMI: 0
2 fanspeed WMI2 error: 0
2 fanspeed WMI2: 0
2 fanspeed WMI3 error: 0
2 fanspeed WMI3: 1900
powermode access method: 3
powermode error: 0
powermode: 2
powermode EC error: 0
powermode EC: 98
powermode ACPI error: -5
powermode ACPI: 0
powermode WMI error: 0
powermode WMI: 2
has custom powermode: 1
ACPI rapidcharge error: 0
ACPI rapidcharge: 1
WMI backlight 2 state: 0
WMI backlight 3 state: 0
WMI light IO port: 1
WMI light Y logo/lid: 1
EC minifancurve feature enabled: 1
EC minifancurve on cool: false
EC lockfancontroller error: -1
EC lockfancontroller: error
fanfullspeed error: -22
fanfullspeed: 0
fanfullspeed EC error: -1
fanfullspeed EC: 0
EC fan curve current point id: 10
EC fan curve points size: 10
Current fan curve in hardware:
rpm1|rpm2|acceleration|deceleration|cpu_min_temp|cpu_max_temp|gpu_min_temp|gpu_max_temp|ic_min_temp|ic_max_temp
0	 0	 0	 0	 0	 50	 0	 50	 0	 0
0	 0	 0	 0	 0	 55	 0	 55	 0	 0
0	 0	 0	 0	 0	 60	 0	 60	 0	 0
1000	 1000	 0	 0	 0	 65	 0	 65	 0	 0
1200	 1200	 0	 0	 0	 70	 0	 70	 0	 0
1600	 1600	 0	 0	 0	 78	 0	 78	 0	 0
1800	 1800	 0	 0	 0	 83	 0	 83	 0	 0
2600	 2600	 0	 0	 0	 88	 0	 88	 0	 0
3000	 3000	 0	 0	 0	 127	 0	 127	 0	 0
4000	 4000	 0	 0	 0	 100	 0	 19	 0	 0
=====================
Current fan curve in hardware (WMI; might be empty)
rpm1|rpm2|acceleration|deceleration|cpu_min_temp|cpu_max_temp|gpu_min_temp|gpu_max_temp|ic_min_temp|ic_max_temp
=====================
```

johnfanv2, 2024-11-16:
@st0nie Could we detect and switch automatically?

st0nie, 2024-11-18:
https://github.com/johnfanv2/LenovoLegionLinux/commit/e0a361f53f3816256e308d26cee636c1294f7925

This commit may helps

closed 2024-12-18 by random2907