johnfanv2/LenovoLegionLinux #281: [BUG] module not buildable under kernel 6.11.5

thomasgoirand, 2024-11-16:
**Problem Description**
Installing the new Debian kernel from backports, linux-image-6.11.5+bpo-amd64, the dkms module failed to build with the below log:

make: Entering directory '/usr/src/linux-headers-6.11.5+bpo-amd64'
  CC [M]  /var/lib/dkms/legion-laptop/0.0.16/build/legion-laptop.o
/var/lib/dkms/legion-laptop/0.0.16/build/legion-laptop.c:6118:19: error: initialization of ‘void (*)(struct platform_device *)’ from incompatible pointer type ‘int (*)(struct platform_device *)’ [-Werror=incompatible-pointer-types]
 6118 |         .remove = legion_remove,
      |                   ^~~~~~~~~~~~~
/var/lib/dkms/legion-laptop/0.0.16/build/legion-laptop.c:6118:19: note: (near initialization for ‘legion_driver.<anonymous>.remove’)
/var/lib/dkms/legion-laptop/0.0.16/build/legion-laptop.c:1787:16: warning: ‘ecram_memoryio_write’ defined but not used [-Wunused-function]
 1787 | static ssize_t ecram_memoryio_write(const struct ecram_memoryio *ec_memoryio,
      |                ^~~~~~~~~~~~~~~~~~~~
cc1: some warnings being treated as errors

**Model and Debug Info**

Installation type: Debian package backported from unstable
Distribution: Debian 12.8

FYI, I'm the package maintainer in Debian for the lenovolegionlinux package. Please help me fix this in Debian.

thomasgoirand, 2024-11-16:
Please note that I tried upgrading to 0.0.18, with the unfortunate same result.

st0nie, 2024-11-18:
taged https://github.com/johnfanv2/LenovoLegionLinux/releases/tag/v0.0.19

closed 2024-11-18 by st0nie