johnfanv2/LenovoLegionLinux #283: [BUG] Fedora 41 dkms-LenovoLegionLinux package can't activate with dkms

class1c-j, 2024-11-19:
**Problem Description**

When installing the driver with the package from https://copr.fedorainfracloud.org/coprs/mrduarte/LenovoLegionLinux/, the module is not correctly loaded into the kernel and I get the following errors:

when running: systemctl status dkms.service
```
× dkms.service - Builds and install new kernel modules through DKMS
     Loaded: loaded (/usr/lib/systemd/system/dkms.service; enabled; preset: enabled)
    Drop-In: /usr/lib/systemd/system/service.d
             └─10-timeout-abort.conf, 50-keep-warm.conf
     Active: failed (Result: exit-code) since Tue 2024-11-19 19:38:57 WET; 13min ago
 Invocation: 0dc48c96536244638ea82d4a27991041
       Docs: man:dkms(8)
    Process: 1132 ExecStart=/usr/sbin/dkms autoinstall --verbose --kernelver 6.11.8-300.fc41.x86_64 (code=exited, status=11)
   Main PID: 1132 (code=exited, status=11)
   Mem peak: 157.3M
        CPU: 1.017s

nov 19 19:38:57 legion-fedora dkms[1389]: (bad exit status: 2)
nov 19 19:38:57 legion-fedora dkms[1775]: Error! Bad return status for module build on kernel: 6.11.8-300.fc41.x86_64 (x86_64)
nov 19 19:38:57 legion-fedora dkms[1775]: Consult /var/lib/dkms/LenovoLegionLinux/0.0.18/build/make.log for more information.
nov 19 19:38:57 legion-fedora dkms[1132]: Autoinstall on 6.11.8-300.fc41.x86_64 failed for module(s) LenovoLegionLinux(10).
nov 19 19:38:57 legion-fedora dkms[1776]: Error! One or more modules failed to install during autoinstall.
nov 19 19:38:57 legion-fedora dkms[1776]: Refer to previous errors for more information.
nov 19 19:38:57 legion-fedora systemd[1]: dkms.service: Main process exited, code=exited, status=11/n/a
nov 19 19:38:57 legion-fedora systemd[1]: dkms.service: Failed with result 'exit-code'.
nov 19 19:38:57 legion-fedora systemd[1]: Failed to start dkms.service - Builds and install new kernel modules through DKMS.
nov 19 19:38:57 legion-fedora systemd[1]: dkms.service: Consumed 1.017s CPU time, 157.3M memory peak.
```

when running: cat /var/lib/dkms/LenovoLegionLinux/0.0.18/build/make.log 
```
DKMS make.log for LenovoLegionLinux-0.0.18 for kernel 6.11.8-300.fc41.x86_64 (x86_64)
ter 19 nov 2024 19:41:53 WET
make -C /lib/modules/6.11.8-300.fc41.x86_64/build M=/var/lib/dkms/LenovoLegionLinux/0.0.18/build modules
make[1]: Entering directory '/usr/src/kernels/6.11.8-300.fc41.x86_64'
  CC [M]  /var/lib/dkms/LenovoLegionLinux/0.0.18/build/legion-laptop.o
/var/lib/dkms/LenovoLegionLinux/0.0.18/build/legion-laptop.c:6147:19: error: initialization of ‘void (*)(struct platform_device *)’ from incompatible pointer type ‘int (*)(struct platform_device *)’ [-Wincompatible-pointer-types]
 6147 |         .remove = legion_remove,
      |                   ^~~~~~~~~~~~~
/var/lib/dkms/LenovoLegionLinux/0.0.18/build/legion-laptop.c:6147:19: note: (near initialization for ‘legion_driver.<anonymous>.remove’)
/var/lib/dkms/LenovoLegionLinux/0.0.18/build/legion-laptop.c:1816:16: warning: ‘ecram_memoryio_write’ defined but not used [-Wunused-function]
 1816 | static ssize_t ecram_memoryio_write(const struct ecram_memoryio *ec_memoryio,
      |                ^~~~~~~~~~~~~~~~~~~~
make[3]: *** [scripts/Makefile.build:244: /var/lib/dkms/LenovoLegionLinux/0.0.18/build/legion-laptop.o] Error 1
make[2]: *** [/usr/src/kernels/6.11.8-300.fc41.x86_64/Makefile:1966: /var/lib/dkms/LenovoLegionLinux/0.0.18/build] Error 2
make[1]: *** [Makefile:236: __sub-make] Error 2
make[1]: Leaving directory '/usr/src/kernels/6.11.8-300.fc41.x86_64'
make: *** [Makefile:13: all] Error 2

```

**Model and Debug Info**

Installation type: package manager (dnf and copr repo)
Distribution: Fedora 41
Model name: Legion Slim 5 16" 2023
CPU model: 13th Gen Intel(R) Core(TM) i7-13z
GPU model: NVIDIA GeForce RTX 4070 Max-Q e
Keyboard backlight: single color with off/medium/bright
Light in lid or logo: no
Light at IO-Ports at back: yes

Output of `sudo dmidecode -t system`. Please remove Serial Number and UUID for privacy:
```text
# dmidecode 3.6
Getting SMBIOS data from sysfs.
SMBIOS 3.4 present.

Handle 0x0001, DMI type 1, 27 bytes
System Information
	Manufacturer: LENOVO
	Product Name: 82YA
	Version: Legion Slim 5 16IRH8
	Serial Number:
	UUID: 
	Wake-up Type: Power Switch
	SKU Number: LENOVO_MT_82YA_BU_idea_FM_Legion Slim 5 16IRH8
	Family: Legion Slim 5 16IRH8

Handle 0x0024, DMI type 12, 5 bytes
System Configuration Options
	Option 1: ConfigOptions1
	Option 2: ConfigOptions2
	Option 3: ConfigOptions3

Handle 0x0027, DMI type 15, 29 bytes
System Event Log
	Area Length: 0 bytes
	Header Start Offset: 0x0000
	Header Length: 8192 bytes
	Data Start Offset: 0x2000
	Access Method: General-purpose non-volatile data functions
	Access Address: 0x0000
	Status: Valid, Not Full
	Change Token: 0x12345678
	Header Format: OEM-specific
	Supported Log Type Descriptors: 3
	Descriptor 1: POST memory resize
	Data Format 1: None
	Descriptor 2: POST error
	Data Format 2: POST results bitmap
	Descriptor 3: Log area reset/cleared
	Data Format 3: None

Handle 0x0034, DMI type 32, 11 bytes
System Boot Information
	Status: No errors detected

```

Output of `sudo dmidecode -t bios`:
```text
# dmidecode 3.6
Getting SMBIOS data from sysfs.
SMBIOS 3.4 present.

Handle 0x0000, DMI type 0, 26 bytes
BIOS Information
	Vendor: LENOVO
	Version: M2CN25WW
	Release Date: 05/16/2023
	Address: 0xE0000
	Runtime Size: 128 kB
	ROM Size: 16 MB
	Characteristics:
		PCI is supported
		BIOS is upgradeable
		BIOS shadowing is allowed
		Boot from CD is supported
		Selectable boot is supported
		EDD is supported
		Japanese floppy for NEC 9800 1.2 MB is supported (int 13h)
		Japanese floppy for Toshiba 1.2 MB is supported (int 13h)
		5.25"/360 kB floppy services are supported (int 13h)
		5.25"/1.2 MB floppy services are supported (int 13h)
		3.5"/720 kB floppy services are supported (int 13h)
		3.5"/2.88 MB floppy services are supported (int 13h)
		8042 keyboard services are supported (int 9h)
		CGA/mono video services are supported (int 10h)
		ACPI is supported
		USB legacy is supported
		BIOS boot specification is supported
		Targeted content distribution is supported
		UEFI is supported
	BIOS Revision: 2.25
	Firmware Revision: 2.25

Handle 0x0025, DMI type 13, 22 bytes
BIOS Language Information
	Language Description Format: Long
	Installable Languages: 8
		en|US|iso8859-1,0
		fr|FR|iso8859-1,0
		zh|TW|unicode,0
		ja|JP|unicode,0
		it|IT|iso8859-1,0
		es|ES|iso8859-1,0
		de|DE|iso8859-1,0
		pt|PT|iso8859-1,0
	Currently Installed Language: en|US|iso8859-1,0
```

Output of `sudo cat /sys/kernel/debug/legion/fancurve`:
```text
cat: /sys/kernel/debug/legion/fancurve: No such file or directory
```

st0nie, 2024-11-21:
tag v0.0.19 may fix this

RodrigoPrestes, 2024-11-21:
What do we need to do to trigger a build on copr?
https://github.com/johnfanv2/LenovoLegionLinux/issues/273 might also be fixed with this.

st0nie, 2024-11-23:
CC @MrDuartePT

MrDuartePT, 2024-11-23:
I did forget to do a new Tag, thanks @st0nie. I will update the spec and rebuild the package in a few minutes

MrDuartePT, 2024-11-23:
@class1c-j @RodrigoPrestes @emilf the package finish build now. If it solve close the issue please
[image]

class1c-j, 2024-11-23:
This fixes the build error and the module is correctly loaded now. Thanks @MrDuartePT and everyone for the quick solution!

closed 2024-11-23 by class1c-j