johnfanv2/LenovoLegionLinux #287: [BUG] Python scripts no longer work

anasbouzid, 2024-12-04:
**Problem Description**
After [this commit](https://github.com/johnfanv2/LenovoLegionLinux/commit/08ff2a6fc4f4df1308756e061f782f5572c98b46), I'm getting the following error message when executing the Python GUI `legion_gui.py` then trying to set or read a profile:
```text
Traceback (most recent call last):
  File "~/LenovoLegionLinux/python/legion_linux/legion_linux/legion_gui.py", line 869, in on_read_fan_curve_from_hw
    self.model.read_fancurve_from_hw()
  File "~/LenovoLegionLinux/python/legion_linux/legion_linux/../legion_linux/legion.py", line 1486, in read_fancurve_from_hw
    self.fan_curve = self.fancurve_io.read_fan_curve()
  File "~/LenovoLegionLinux/python/legion_linux/legion_linux/../legion_linux/legion.py", line 960, in read_fan_curve
    fan1_speed = self.get_fan_1_speed_rpm(point_id)
  File "~/LenovoLegionLinux/python/legion_linux/legion_linux/../legion_linux/legion.py", line 868, in get_fan_1_speed_rpm
    return round(self.get_fan_1_speed_pwm(point_id)/255.0*self.get_fan_1_max_rpm(), ndigits=2)
  File "~/LenovoLegionLinux/python/legion_linux/legion_linux/../legion_linux/legion.py", line 795, in get_fan_1_max_rpm
    return int(self._read_file(file_path))
  File "~/LenovoLegionLinux/python/legion_linux/legion_linux/../legion_linux/legion.py", line 774, in _read_file
    with open(file_path, "r", encoding=DEFAULT_ENCODING) as filepointer:
FileNotFoundError: [Errno 2] No such file or directory: '/sys/module/legion_laptop/drivers/platform:legion/PNP0C09:00/hwmon/hwmon4/fan1_max'
```
This is the list of all files inside `/sys/module/legion_laptop/drivers/platform:legion/PNP0C09:00/hwmon/hwmon4/`:
```
~ ls /sys/module/legion_laptop/drivers/platform:legion/PNP0C09:00/hwmon/hwmon4/
auto_points_size             pwm1_auto_point5_temp_hyst   pwm2_auto_point6_pwm
device                       pwm1_auto_point6_accel       pwm2_auto_point6_temp
fan1_input                   pwm1_auto_point6_decel       pwm2_auto_point6_temp_hyst
fan1_label                   pwm1_auto_point6_pwm         pwm2_auto_point7_pwm
fan1_target                  pwm1_auto_point6_temp        pwm2_auto_point7_temp
fan2_input                   pwm1_auto_point6_temp_hyst   pwm2_auto_point7_temp_hyst
fan2_label                   pwm1_auto_point7_accel       pwm2_auto_point8_pwm
fan2_target                  pwm1_auto_point7_decel       pwm2_auto_point8_temp
minifancurve                 pwm1_auto_point7_pwm         pwm2_auto_point8_temp_hyst
name                         pwm1_auto_point7_temp        pwm2_auto_point9_pwm
power                        pwm1_auto_point7_temp_hyst   pwm2_auto_point9_temp
pwm1_auto_point10_accel      pwm1_auto_point8_accel       pwm2_auto_point9_temp_hyst
pwm1_auto_point10_decel      pwm1_auto_point8_decel       pwm3_auto_point10_temp
pwm1_auto_point10_pwm        pwm1_auto_point8_pwm         pwm3_auto_point10_temp_hyst
pwm1_auto_point10_temp       pwm1_auto_point8_temp        pwm3_auto_point1_temp
pwm1_auto_point10_temp_hyst  pwm1_auto_point8_temp_hyst   pwm3_auto_point1_temp_hyst
pwm1_auto_point1_accel       pwm1_auto_point9_accel       pwm3_auto_point2_temp
pwm1_auto_point1_decel       pwm1_auto_point9_decel       pwm3_auto_point2_temp_hyst
pwm1_auto_point1_pwm         pwm1_auto_point9_pwm         pwm3_auto_point3_temp
pwm1_auto_point1_temp        pwm1_auto_point9_temp        pwm3_auto_point3_temp_hyst
pwm1_auto_point1_temp_hyst   pwm1_auto_point9_temp_hyst   pwm3_auto_point4_temp
pwm1_auto_point2_accel       pwm1_mode                    pwm3_auto_point4_temp_hyst
pwm1_auto_point2_decel       pwm2_auto_point10_pwm        pwm3_auto_point5_temp
pwm1_auto_point2_pwm         pwm2_auto_point10_temp       pwm3_auto_point5_temp_hyst
pwm1_auto_point2_temp        pwm2_auto_point10_temp_hyst  pwm3_auto_point6_temp
pwm1_auto_point2_temp_hyst   pwm2_auto_point1_pwm         pwm3_auto_point6_temp_hyst
pwm1_auto_point3_accel       pwm2_auto_point1_temp        pwm3_auto_point7_temp
pwm1_auto_point3_decel       pwm2_auto_point1_temp_hyst   pwm3_auto_point7_temp_hyst
pwm1_auto_point3_pwm         pwm2_auto_point2_pwm         pwm3_auto_point8_temp
pwm1_auto_point3_temp        pwm2_auto_point2_temp        pwm3_auto_point8_temp_hyst
pwm1_auto_point3_temp_hyst   pwm2_auto_point2_temp_hyst   pwm3_auto_point9_temp
pwm1_auto_point4_accel       pwm2_auto_point3_pwm         pwm3_auto_point9_temp_hyst
pwm1_auto_point4_decel       pwm2_auto_point3_temp        subsystem
pwm1_auto_point4_pwm         pwm2_auto_point3_temp_hyst   temp1_input
pwm1_auto_point4_temp        pwm2_auto_point4_pwm         temp1_label
pwm1_auto_point4_temp_hyst   pwm2_auto_point4_temp        temp2_input
pwm1_auto_point5_accel       pwm2_auto_point4_temp_hyst   temp2_label
pwm1_auto_point5_decel       pwm2_auto_point5_pwm         temp3_input
pwm1_auto_point5_pwm         pwm2_auto_point5_temp        temp3_label
pwm1_auto_point5_temp        pwm2_auto_point5_temp_hyst   uevent
``` 

**Model and Debug Info**

Installation type: via package manager,
Distribution: KDE neon 5.27, ubuntu 22.04
Model name: 16ACH6H
CPU model: AMD Ryzen 7 5800H
GPU model: NVIDIA RTX 3700
[... 152 lines cut ...]
EC minifancurve feature enabled: 1
EC minifancurve on cool: false
EC lockfancontroller error: 0
EC lockfancontroller: false
fanfullspeed error: 0
fanfullspeed: 0
fanfullspeed EC error: 0
fanfullspeed EC: 0
EC fan curve current point id: 0
EC fan curve points size: 8
Current fan curve in hardware:
rpm1|rpm2|acceleration|deceleration|cpu_min_temp|cpu_max_temp|gpu_min_temp|gpu_max_temp|ic_min_temp|ic_max_temp
0        0       2       2       0       127     0       47      0       48
1100     1100    2       2       127     127     47      52      48      52
1500     1500    2       2       127     127     52      58      52      56
2100     2100    2       2       127     127     58      66      56      127
2300     2300    2       2       127     127     66      70      127     127
2600     2600    2       2       127     127     70      75      127     127
2800     2800    2       2       127     127     75      80      127     127
3500     3500    2       2       127     127     80      127     127     127
=====================
Current fan curve in hardware (WMI; might be empty)
rpm1|rpm2|acceleration|deceleration|cpu_min_temp|cpu_max_temp|gpu_min_temp|gpu_max_temp|ic_min_temp|ic_max_temp
=====================
```

johnfanv2, 2024-12-04:
Sorry, we updated the kernel and the GUI at the same time but did not make the GUI compatible with lower level version kernels. We need to make a breaking change to follow official kernel standard to eventually merge it into main line kerne. Please just update the kernel module.

Which package from which source did you use to install it? It should be correct in the package sources. If you installed a dkms package for the kernel module please also try to update this.

anasbouzid, 2024-12-04:
I was using v0.0.15, then I upgraded to v0.0.18 via the .deb file [from this link](https://packages.debian.org/sid/amd64/lenovolegionlinux-dkms/download) (I followed the Debian repo link in the README page). I tried to install it via DKMS but no luck, no error when building but when I reboot, the module is not loaded.

anasbouzid, 2024-12-04:
I'm currently using an old commit, everything works perfectly. Thank you so much for your work.

johnfanv2, 2024-12-08:
Hey, do you have problems with the newest version and fixed it by using an old commit? Or did you had problems with an old commit?

anasbouzid, 2024-12-10:
I had problems with the newest version and fixed it by using an old commit.

closed 2026-09-11 by github-actions[bot]