johnfanv2/LenovoLegionLinux #288: Upstream Lenovo Legion Linux driver

pm4rcin, 2024-12-18:
Hi, it seems like Derek J. Clark sent patches to upstream parts of the driver to mainline. It's not everything but it's worth looking at and maybe even contacting him since I don't know if he is aware of this repository existing because he made the patches based on Legion Go console as noted in thread. [Here's the thread](https://lkml.org/lkml/2024/12/17/1611) on LKML. What do you guys think?

MartinVonReichenberg, 2025-01-13:
Here are the download links pointing to the actual driver:
[Lenovo Legion WMI](https://github.com/adnanpri/legion-wmi)
[Lenovo Legion WMI - DKMS](https://github.com/null-dev/lenovo-legion-wmi-dkms)

chennin, 2025-01-14:
> Here are the download links pointing to the actual driver: [Lenovo Legion WMI](https://github.com/adnanpri/legion-wmi) [Lenovo Legion WMI - DKMS](https://github.com/null-dev/lenovo-legion-wmi-dkms)

Those repos are last updated 3 and 4 years ago, were those the right links? Those don't look like "upstream" and the first one even says "THIS DRIVER DOES NOT WORK".

MrDuartePT, 2025-02-04:
The driver will be upstream when it ready, @johnfanv2 did send the driver to kernel maling list for feedback and we need to rewrite a couple of thing before being considered "upstream-able":
https://lore.kernel.org/lkml/20241226001828.423658-1-johnfanv2@gmail.com/

ryanabx, 2025-02-04:
It might also be good to make sure the driver works for newer legion devices, considering all the issues asking for support and [this thread](https://github.com/johnfanv2/LenovoLegionLinux/issues/46)

I'm not sure but I think the other drive Derek Clark made covered modern legions as well, so inspiration could be taken from that.

st0nie, 2025-05-12:
https://github.com/torvalds/linux/commit/02c6e43397c39edd0c172859bf8c851b46be09a8
Recently the linux kernel added support for platform_profile_cycle for Fn+Q for ideapad.

ryanabx, 2025-05-12:
> [torvalds/linux@02c6e43](https://github.com/torvalds/linux/commit/02c6e43397c39edd0c172859bf8c851b46be09a8) Recently the linux kernel added support for platform_profile_cycle for Fn+Q for ideapad.

That's cool -- does it sync up with stuff like power-profiles-daemon? If I Fn+Q will it desync my platform profile with the power profile from power-profiles-daemon?

ryanabx, 2025-05-21:
Did some research -- unsure if it will desync from ppd, but given that it's using the standard platform profile interface, it should sync up nicely for when the new [lenovo wmi drivers](https://lore.kernel.org/platform-driver-x86/637c0f0e-3858-43bd-b122-ff52a5580077@gmx.de/T/#mefef63e2bcdcbddd77a9bde3128dcbdaed0589e4) land upstream (should be really soon!)

closed 2026-09-11 by github-actions[bot]