johnfanv2/LenovoLegionLinux #289: Lenovo Legion T5 28IMB05 Support?

CE1CECL, 2024-12-21:
Ref: https://github.com/BartoszCichecki/LenovoLegionToolkit/issues/69, https://github.com/ViRb3/LenovoController/issues/4 & https://github.com/4JX/L5P-Keyboard-RGB/issues/61

This desktop finally got support for zone 3 & 4 with OpenRGB (https://gitlab.com/CalcProgrammer1/OpenRGB/-/issues/2999 & https://gitlab.com/CalcProgrammer1/OpenRGB/-/merge_requests/1602)
But it still is missing the `wmic /namespace:\\root\wmi path LENOVO_GAMEZONE_DATA where "InstanceName='ACPI\\PNP0C14\\GamingWMI_0'" call /?` part: 
```

Method execution operations.
USAGE:

CALL <method name> [<actual paramlist>]
NOTE: <actual paramlist> ::= <actual param> | <actual param>,  <actual paramlist>

The following verb(s)/method(s) are available:

Call			[ In/Out ]Params&type			Status
====			=====================			======
GetFanList              [OUT]Data(UINT32)                   	Implemented    

GetFanSpeed             [IN ]idx(UINT32)                    	Implemented    

			[OUT]Data(UINT32)                   	

GetLED                  [IN ]idx(UINT32)                    	Implemented    

			[OUT]Data(UINT32)                   	

GetMemoryOCInfo         [OUT]Data(UINT32)                   	Implemented    

GetOCSetmode            [OUT]Data(UINT32)                   	Implemented    

GetProductInfo          [OUT]Data(UINT32)                   	Implemented    

GetSensorTemperature    [IN ]idx(UINT32)                    	Implemented    

			[OUT]Data(UINT32)                   	

GetSmartFanMode         [OUT]Data(UINT32)                   	Implemented    

GetSupportThermalMode   [OUT]Data(UINT32)                   	Implemented    

GetVersion              [OUT]Data(UINT32)                   	Implemented    

IsRestoreOCValue        [IN ]idx(UINT32)                    	Implemented    

			[OUT]Data(UINT32)                   	

IsSupportBIOSLED        [OUT]Data(UINT32)                   	Implemented    

IsSupportCpuOC          [OUT]Data(UINT32)                   	Implemented    

IsSupportEnergyStar     [OUT]Data(UINT32)                   	Implemented    

IsSupportGpuOC          [OUT]Data(UINT32)                   	Implemented    

IsSupportLightingFeature[OUT]Data(UINT32)                   	Implemented    

IsSupportSmartFan       [OUT]Data(UINT32)                   	Implemented    

IsSupportWaterCooling   [OUT]Data(UINT32)                   	Implemented    

SetLED                  [IN ]Data(UINT32)                   	Implemented    

SetOCSetmode            [IN ]Data(UINT32)                   	Implemented    

SetSmartFanMode         [IN ]Data(UINT32)                   	Implemented    
```
Anything I can do for this 2020 tower desktop to support RGB in anything BUT Windows 10, almost 5 years later after purchase? Zones 1-2 seem to be through ACPI/WMI somehow.

CE1CECL, 2026-09-03:
> This issue has been automatically marked as stale because it has had no activity for 90 days. **If this issue is still affecting you, please leave any comment** (for example, "bump") and it will stay open. It will be closed in 7 days if no further activity occurs.

bump

st0nie, 2026-09-14:
Closing as out of scope: the T5 28IMB05 is a desktop without the laptop EC/WMI interface this driver is built around. As you found, OpenRGB covers its lighting; fan control on the desktop is handled by the motherboard firmware, not this EC interface.

closed 2026-09-14 by st0nie