johnfanv2/LenovoLegionLinux #292: [BUG] [Fedora 41] [v0.0.19] Fn + Q key and power profiles do not work properly.

Allephh, 2025-01-10:
**Problem Description**
Using the Fn + Q key to change power profiles does not change the power profiles in the Quick Menu and has little impact on gaming performance.
At the same time, despite changing power profiles via the Quick Menu, the behavior of the LED on the laptop's power button is strange and have a delay when altering the Power Modes.
In both situations, the impact on performance and temperature is low and does not reflect that they are being used, unlike previous installations such as Fedora 39 and 40.
I believe the problem is related to this:
https://fedoraproject.org/wiki/Changes/TunedAsTheDefaultPowerProfileManagementDaemon

https://github.com/user-attachments/assets/afc211b5-0b89-43ce-bc56-ff68df8c6a54

**Hardware Information:**
- **Hardware Model:**                     Lenovo IdeaPad Gaming 3 15ARH7
- **Memory:**                                    16.0 GiB
- **Processor:**                                  AMD Ryzen™ 7 6800H with Radeon™ Graphics × 16
- **Graphics:**                                    AMD Radeon™ 680M and NVDIA RTX 3050
- **Disk Capacity:**                            512 GB

**Software Information:**
- **Firmware Version:**                     JNCN51WW(V2.11)
- **OS Name:**                                   Fedora Linux 41 (Workstation Edition)       
- **OS Type:**                                     64-bit
- **GNOME Version:**                       47
- **Windowing System:**                  Wayland
- **Kernel Version:**                          Linux 6.12.7-200.fc41.x86_64

MartinVonReichenberg, 2025-01-11:
_That never worked for me on Linux . . ._

MrDuartePT, 2025-01-13:
That is a fedora change that we have no control about and the driver already uses `/sys/firmware/acpi/platform_profile` that is the default that exposes the powerprofile for both ppd and tuned.

About the quick menu that is a tuned, gnome or fedora bug and we can't really do nothing about that, since when the power profile is changed by pressing FN+Q the value inside `/sys/firmware/acpi/platform_profile` also get change (I use TLP and it reports the change).

closed 2025-01-13 by MrDuartePT

MartinVonReichenberg, 2025-01-13:
Hello @Allephh,

You can try this/these:
[Lenovo Legion WMI](https://github.com/adnanpri/legion-wmi)
[Lenovo Legion WMI - DKMS](https://github.com/null-dev/lenovo-legion-wmi-dkms)

MrDuartePT, 2025-01-13:
> Hello @Allephh,
> 
> You can try this/these: [Lenovo Legion WMI](https://github.com/adnanpri/legion-wmi) [Lenovo Legion WMI - DKMS](https://github.com/null-dev/lenovo-legion-wmi-dkms)

This is outdated on my machine in the latest Nvidia-driver I can get full performance (more 80w) without this and if you use our driver in custom mode you can get the max tdp that vbios allows.