johnfanv2/LenovoLegionLinux #293: [BUG] No Sound on kernel 6.12.5 and higher for 16ARX8H (and probably others)

sammilucia, 2025-01-21:
**Problem Description**
<Please describe your problem here>
Has the bug occurred following steps described in the README? Yes
What steps did you take?
- Bifurcated with Fedora and nobara kernels
- Kernel 6.12.4 nobara/Fedora kernels work correctly, and no sound on 6.12.5
- Tried previous TAS2781 suspend fixes, no change
What commands did you enter?
What was the (full) error you received and what command or action did lead to it?
- I'll reboot into broken kernel 6.12.5 and update this issue with the outputs in a sec

**Model and Debug Info**

Installation type: (problem occurs with or without DKMS) https://copr.fedorainfracloud.org/coprs/mrduarte/LenovoLegionLinux
Distribution: Fedora 41, nobara 41 (also broken on cachyOS as at 19-Jan-25 so suspect mainline kernel)
Model name: 16ARX8H Lenovo Legion Pro 7 Gen 8 (2024)
CPU model: AMD Ryzen 9 7945HX
GPU model: Nvidia RTX 4090
Keyboard backlight: RGB per key
Light in lid or logo: no
Light at IO-Ports at back: no

Output of `sudo dmidecode -t system`. Please remove Serial Number and UUID for privacy:
```text
# dmidecode 3.6
Getting SMBIOS data from sysfs.
SMBIOS 3.4.0 present.

Handle 0x0001, DMI type 1, 27 bytes
System Information
	Manufacturer: LENOVO
	Product Name: 82WS
	Version: Legion Pro 7 16ARX8H
	Serial Number: PF4YZPGK
	UUID: 6bf9f14b-0e39-4e96-b7c5-fc5ceed02c01
	Wake-up Type: Power Switch
	SKU Number: LENOVO_MT_82WS_BU_idea_FM_Legion Pro 7 16ARX8H
	Family: Legion Pro 7 16ARX8H

Handle 0x0017, DMI type 12, 5 bytes
System Configuration Options
	Option 1: String1 for Type12 Equipment Manufacturer
	Option 2: String2 for Type12 Equipment Manufacturer
	Option 3: String3 for Type12 Equipment Manufacturer
	Option 4: String4 for Type12 Equipment Manufacturer

Handle 0x0026, DMI type 32, 20 bytes
System Boot Information
	Status: No errors detected
```

Output of `sudo dmidecode -t bios`:
```text
# dmidecode 3.6
Getting SMBIOS data from sysfs.
SMBIOS 3.4.0 present.

Handle 0x0000, DMI type 0, 26 bytes
BIOS Information
	Vendor: LENOVO
	Version: LPCN59WW
	Release Date: 11/25/2024
	Address: 0xE0000
	Runtime Size: 128 kB
	ROM Size: 32 MB
	Characteristics:
		PCI is supported
		PNP is supported
		BIOS is upgradeable
		BIOS shadowing is allowed
		Boot from CD is supported
		Selectable boot is supported
		EDD is supported
		ACPI is supported
		USB legacy is supported
		BIOS boot specification is supported
		Function key-initiated network boot is supported
		Targeted content distribution is supported
		UEFI is supported
	BIOS Revision: 1.59
	Firmware Revision: 1.59

Handle 0x0018, DMI type 13, 22 bytes
BIOS Language Information
	Language Description Format: Long
	Installable Languages: 4
		en|US|iso8859-1
		fr|FR|iso8859-1
		ja|JP|unicode
		zh|TW|unicode
	Currently Installed Language: en|US|iso8859-1
```

Output of `sudo cat /sys/kernel/debug/legion/fancurve`:
```text
EC Chip ID: 5507
EC Chip Version: 2a4
legion_laptop features: fancurve powermode platformprofile platformprofilenotify minifancurve fancurve_pmw_speed fancurve_rpm_speed
legion_laptop ec_readonly: 0
ACPI CFG error: 0
ACPI CFG: 2081289482
temperature access method: 5
CPU temperature error: 0
CPU temperature: 68
CPU temperature EC error: 0
CPU temperature EC: 0
CPU temperature ACPI error: 0
CPU temperature ACPI: 68
CPU temperature WMI error: 0
CPU temperature WMI: 0
CPU temperature WMI2 error: 0
CPU temperature WMI2: 0
CPU temperature WMI3 error: 0
CPU temperature WMI3: 68
GPU temperature error: 0
GPU temperature: 51
GPU temperature EC error: 0
GPU temperature EC: 0
GPU temperature ACPI error: 0
GPU temperature ACPI: 51
GPU temperature WMI error: 0
GPU temperature WMI: 0
GPU temperature WMI2 error: 0
GPU temperature WMI2: 0
GPU temperature WMI3 error: 0
GPU temperature WMI3: 51
fan speed access method: 5
1 fanspeed error: 0
1 fanspeed: 2800
1 fanspeed EC error: 0
1 fanspeed EC: 48
1 fanspeed ACPI error: 0
1 fanspeed ACPI: 2800
1 fanspeed WMI error: 0
1 fanspeed WMI: 0
1 fanspeed WMI2 error: 0
1 fanspeed WMI2: 0
1 fanspeed WMI3 error: 0
1 fanspeed WMI3: 2800
2 fanspeed error: 0
2 fanspeed: 2800
2 fanspeed EC error: 0
2 fanspeed EC: 0
2 fanspeed ACPI error: 0
2 fanspeed ACPI: 2800
2 fanspeed WMI error: 0
2 fanspeed WMI: 0
2 fanspeed WMI2 error: 0
2 fanspeed WMI2: 0
2 fanspeed WMI3 error: 0
2 fanspeed WMI3: 2800
powermode access method: 3
powermode error: 0
powermode: 3
powermode EC error: 0
powermode EC: 1
powermode ACPI error: -5
powermode ACPI: 0
powermode WMI error: 0
powermode WMI: 3
has custom powermode: 1
ACPI rapidcharge error: 0
ACPI rapidcharge: 1
WMI backlight 2 state: 0
WMI backlight 3 state: 0
WMI light IO port: 0
WMI light Y logo/lid: 0
EC minifancurve feature enabled: 0
EC minifancurve on cool: error
EC lockfancontroller error: 0
EC lockfancontroller: false
fanfullspeed error: 0
fanfullspeed: 0
fanfullspeed EC error: 0
fanfullspeed EC: 0
Max speed for fancurve: 10000
Current fan curve in hardware:
Fan curve current point id: 0
Fan curve points size: 10
u(speed_of_unit)|speed1[u]|speed2[u]|speed1[pwm]|speed2[pwm]|acceleration|deceleration|cpu_min_temp|cpu_max_temp|gpu_min_temp|gpu_max_temp|ic_min_temp|ic_max_temp
1	 0	 0	 0	 0	 0	 0	 0	 0	 0	 0	 0	 0
1	 0	 0	 0	 0	 0	 0	 0	 0	 0	 0	 0	 0
1	 0	 0	 0	 0	 0	 0	 0	 0	 0	 0	 0	 0
1	 0	 0	 0	 0	 0	 0	 0	 0	 0	 0	 0	 0
1	 0	 0	 0	 0	 0	 0	 0	 0	 0	 0	 0	 0
1	 0	 0	 0	 0	 0	 0	 0	 0	 0	 0	 0	 0
1	 0	 0	 0	 0	 0	 0	 0	 0	 0	 0	 0	 0
1	 0	 0	 0	 0	 0	 0	 0	 0	 0	 0	 0	 0
1	 0	 0	 0	 0	 0	 0	 0	 0	 0	 0	 0	 0
1	 0	 0	 0	 0	 0	 0	 0	 0	 0	 0	 0	 0
=====================
Current fan curve in hardware (WMI; might be empty)
Fan curve current point id: 0
Fan curve points size: 10
u(speed_of_unit)|speed1[u]|speed2[u]|speed1[pwm]|speed2[pwm]|acceleration|deceleration|cpu_min_temp|cpu_max_temp|gpu_min_temp|gpu_max_temp|ic_min_temp|ic_max_temp
1	 0	 0	 0	 0	 0	 0	 0	 0	 0	 0	 0	 0
1	 0	 0	 0	 0	 0	 0	 0	 0	 0	 0	 0	 0
1	 0	 0	 0	 0	 0	 0	 0	 0	 0	 0	 0	 0
1	 0	 0	 0	 0	 0	 0	 0	 0	 0	 0	 0	 0
1	 0	 0	 0	 0	 0	 0	 0	 0	 0	 0	 0	 0
1	 0	 0	 0	 0	 0	 0	 0	 0	 0	 0	 0	 0
1	 0	 0	 0	 0	 0	 0	 0	 0	 0	 0	 0	 0
1	 0	 0	 0	 0	 0	 0	 0	 0	 0	 0	 0	 0
1	 0	 0	 0	 0	 0	 0	 0	 0	 0	 0	 0	 0
1	 0	 0	 0	 0	 0	 0	 0	 0	 0	 0	 0	 0
=====================
```
*note DKMS should be loaded but i haven't had a chance to verify since i've been bifurcating this issue

output of `sudo dmesg|rg snd` on **working** kernel 6.12.4
```text
[    4.738575] snd_hda_intel 0000:01:00.1: Disabling MSI
[    4.738585] snd_hda_intel 0000:01:00.1: Handle vga_switcheroo audio client
[    4.761665] snd_hda_codec_realtek hdaudioC1D0: bound i2c-TIAS2781:00 (ops tas2781_hda_comp_ops [snd_hda_scodec_tas2781_i2c])
[    4.762172] snd_hda_codec_realtek hdaudioC1D0: autoconfig for ALC287: line_outs=1 (0x17/0x0/0x0/0x0/0x0) type:speaker
[    4.762174] snd_hda_codec_realtek hdaudioC1D0:    speaker_outs=0 (0x0/0x0/0x0/0x0/0x0)
[    4.762176] snd_hda_codec_realtek hdaudioC1D0:    hp_outs=1 (0x21/0x0/0x0/0x0/0x0)
[    4.762177] snd_hda_codec_realtek hdaudioC1D0:    mono: mono_out=0x0
[    4.762178] snd_hda_codec_realtek hdaudioC1D0:    inputs:
[    4.762179] snd_hda_codec_realtek hdaudioC1D0:      Mic=0x19
[    4.762180] snd_hda_codec_realtek hdaudioC1D0:      Internal Mic=0x12
```

sammilucia, 2025-01-21:
well that's interesting, 6.12.5 kernel dmidecode system and bios were the same as expected but

Output of `sudo cat /sys/kernel/debug/legion/fancurve`:
```text
cat: /sys/kernel/debug/legion/fancurve: No such file or directory
```

output of `sudo dmesg|rg snd`
```text
[    4.278863] snd_hda_intel 0000:01:00.1: Disabling MSI
[    4.278871] snd_hda_intel 0000:01:00.1: Handle vga_switcheroo audio client
[    4.374448] snd_hda_codec_realtek hdaudioC1D0: autoconfig for ALC287: line_outs=1 (0x17/0x0/0x0/0x0/0x0) type:speaker
[    4.374451] snd_hda_codec_realtek hdaudioC1D0:    speaker_outs=0 (0x0/0x0/0x0/0x0/0x0)
[    4.374453] snd_hda_codec_realtek hdaudioC1D0:    hp_outs=1 (0x21/0x0/0x0/0x0/0x0)
[    4.374455] snd_hda_codec_realtek hdaudioC1D0:    mono: mono_out=0x0
[    4.374456] snd_hda_codec_realtek hdaudioC1D0:    inputs:
[    4.374457] snd_hda_codec_realtek hdaudioC1D0:      Mic=0x19
[    4.374458] snd_hda_codec_realtek hdaudioC1D0:      Internal Mic=0x12
```

sammilucia, 2025-01-21:
possible suspect commits from 6.12.5 mainline

```text
commit cccaab2b8d63ef5bd931090c1da1e0b7e99bf293
Author: Charles Keepax <ckeepax@opensource.cirrus.com>
Date:   Wed Oct 16 11:03:44 2024 +0800

    Add some more sidecar amplifier quirks.

sammilucia, 2025-01-22:
i've emailed linux-sound CC Takashi (patch author)

cwright814, 2025-01-25:
@sammilucia Thank you for this report. Finally got sound working on my 16ARX8H when downgrading to kernel 6.12.4.

mqasimca, 2025-01-27:
Hi,
I have the same issue with 16ARX8H, fix https://github.com/mqasimca/CachyOS/blob/master/fix-sound.sh

Gert-dev, 2025-02-01:
For what it's worth, I've had the same problem and have applied the same fix for about half a year now. The issue is likely a kernel bug that was being tracked [here](https://bugzilla.kernel.org/show_bug.cgi?id=208555).

It so happens this issue has been closed today. The best place to report it is thus probably (indeed) the kernel as a new issue report - as [also mentioned by Takashi there](https://bugzilla.kernel.org/show_bug.cgi?id=208555#c911).

Whilst I personally appreciate the follow-up as I have the same problem I'm not sure if this issue tracker is the right place for it though - since I don't think this driver does much for sound, it's all in the kernel upstream.

MrDuartePT, 2025-02-04:
I will close this just because is not a bug with LLL, but nice finding :)

closed 2025-02-04 by MrDuartePT

sammilucia, 2025-02-04:
hi, sorry didn't mean to abandon.

to confirm, it's a problem with upstream kernel as of 6.12.5 due to a patch rearranging Texas Instruments amp enablement. I reported to linux-sound and patch author Takashi Iwai who provided the following patch (confirmed working on my model):

this patch has been brought forward into nobara kernel 6.13.x and likely will to cachyOS as well (if not, just ask them).

```patch
---
 sound/pci/hda/patch_realtek.c | 2 +-
 1 file changed, 1 insertion(+), 1 deletion(-)

diff --git a/sound/pci/hda/patch_realtek.c b/sound/pci/hda/patch_realtek.c
index d3c9ed963588..d36a79a8ecbf 100644
--- a/sound/pci/hda/patch_realtek.c
+++ b/sound/pci/hda/patch_realtek.c
@@ -10918,7 +10918,7 @@ static const struct hda_quirk alc269_fixup_tbl[] = {
        SND_PCI_QUIRK(0x17aa, 0x3869, "Lenovo Yoga7 14IAL7", ALC287_FIXUP_YOGA9_14IAP7_BASS_SPK_PIN),
        HDA_CODEC_QUIRK(0x17aa, 0x386e, "Legion Y9000X 2022 IAH7", ALC287_FIXUP_CS35L41_I2C_2),
        SND_PCI_QUIRK(0x17aa, 0x386e, "Yoga Pro 7 14ARP8", ALC285_FIXUP_SPEAKER2_TO_DAC1),
-       HDA_CODEC_QUIRK(0x17aa, 0x386f, "Legion Pro 7 16ARX8H", ALC287_FIXUP_TAS2781_I2C),
+       HDA_CODEC_QUIRK(0x17aa, 0x38a8, "Legion Pro 7 16ARX8H", ALC287_FIXUP_TAS2781_I2C), /* this must match before PCI SSID 17aa:386f below */
        SND_PCI_QUIRK(0x17aa, 0x386f, "Legion Pro 7i 16IAX7", ALC287_FIXUP_CS35L41_I2C_2),
        SND_PCI_QUIRK(0x17aa, 0x3870, "Lenovo Yoga 7 14ARB7", ALC287_FIXUP_YOGA7_14ARB7_I2C),
        SND_PCI_QUIRK(0x17aa, 0x3877, "Lenovo Legion 7 Slim 16ARHA7", ALC287_FIXUP_CS35L41_I2C_2),
```

ty! issue can remain closed.