johnfanv2/LenovoLegionLinux #296: [BUG] module compile error on latest kernel 6.13.0

mreijnde, 2025-01-26:
**Problem Description**
Building of dkms module fails on latest linux kernel 6.13.0.  The compile error is stated in `make.log` below.

After some googling, it seems like a rename in the latest linux kernel in the used interface is the root cause:
https://github.com/torvalds/linux/commit/e70140ba0d2b1a30467d4af6bcfe761327b9ec95

**Workaround used**
Manual update `/usr/src/lenovolegionlinux-1.0.0/legion-laptop.c`:
change  `.remove_new` to `.remove` at line 6146.

After that the dkms module compiles and installs without problems. Fan settings are showing with at `sudo cat /sys/kernel/debug/legion/fancurve`.

Commands used on arch linux to install dkms manual and create boot image:
`sudo dkms install --no-depmod lenovolegionlinux/1.0.0 -k 6.13.0-2-cachyos`
`sudo mkinitcpio -P`

**Model and Debug Info**

Installation type: dkms (using package manager with package lenovolegionlinux-dkms-git)
Distribution: CachyOS Linux version 6.13.0-2-cachyos
Model name: Lenovo Legion 5 Pro 16ACH6H
CPU model: AMD Ryzen 7 5800H
GPU model: NVIDIA RTX 3070

Output manual dkms build `sudo dkms install --no-depmod lenovolegionlinux/1.0.0 -k 6.13.0-2-cachyos`:
```text
Sign command: /usr/lib/modules/6.13.0-2-cachyos/build/scripts/sign-file
Signing key: /var/lib/dkms/mok.key
Public certificate (MOK): /var/lib/dkms/mok.pub

Cleaning build area... done.
Building module(s)...(bad exit status: 2)
Failed command:
make -j16 KERNELRELEASE=6.13.0-2-cachyos KERNELVERSION=6.13.0-2-cachyos LLVM=1

Error! Bad return status for module build on kernel: 6.13.0-2-cachyos (x86_64)
Consult /var/lib/dkms/lenovolegionlinux/1.0.0/build/make.log for more information.
```

Content of dkms `make.log`:
```text
DKMS make.log for lenovolegionlinux/1.0.0 for kernel 6.13.0-2-cachyos (x86_64)
za 25 jan 2025 23:39:40 CET
make -C /lib/modules/6.13.0-2-cachyos/build M=/var/lib/dkms/lenovolegionlinux/1.0.0/build modules
make[1]: Entering directory '/usr/lib/modules/6.13.0-2-cachyos/build'
make[2]: Entering directory '/var/lib/dkms/lenovolegionlinux/1.0.0/build'
  CC [M]  legion-laptop.o
legion-laptop.c:5683:6: warning: variable 'err' is used uninitialized whenever 'if' condition is true [-Wsometimes-uninitialized]
 5683 |         if (!priv->adev) {
      |             ^~~~~~~~~~~
legion-laptop.c:5709:9: note: uninitialized use occurs here
 5709 |         return err;
      |                ^~~
legion-laptop.c:5683:2: note: remove the 'if' if its condition is always false
 5683 |         if (!priv->adev) {
      |         ^~~~~~~~~~~~~~~~~~
 5684 |                 dev_info(dev, "Could not get ACPI handle\n");
      |                 ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
 5685 |                 goto err_acpi_init;
      |                 ~~~~~~~~~~~~~~~~~~~
 5686 |         }
      |         ~
legion-laptop.c:5677:9: note: initialize the variable 'err' to silence this warning
 5677 |         int err;
      |                ^
      |                 = 0
legion-laptop.c:6146:3: error: field designator 'remove_new' does not refer to any field in type 'struct platform_driver'
 6146 |         .remove_new = legion_remove,
      |         ~^~~~~~~~~~~~~~~~~~~~~~~~~~
1 warning and 1 error generated.
make[4]: *** [/usr/lib/modules/6.13.0-2-cachyos/build/scripts/Makefile.build:194: legion-laptop.o] Error 1
make[3]: *** [/usr/lib/modules/6.13.0-2-cachyos/build/Makefile:1997: .] Error 2
make[2]: *** [/usr/lib/modules/6.13.0-2-cachyos/build/Makefile:251: __sub-make] Error 2
make[2]: Leaving directory '/var/lib/dkms/lenovolegionlinux/1.0.0/build'
make[1]: *** [Makefile:251: __sub-make] Error 2
make[1]: Leaving directory '/usr/lib/modules/6.13.0-2-cachyos/build'
make: *** [Makefile:13: all] Error 2
```

636D, 2025-01-26:
I got this error in NixOS unstable as well

mreijnde, 2025-02-01:
Great, I see that pull request to fix issue has been created and merged: johnfanv2/LenovoLegionLinux#294
Thanks!

closed 2025-02-01 by mreijnde