johnfanv2/LenovoLegionLinux #303: [BUG] module compilation fail on Fedora 41 6.13.* kernels

Korulag, 2025-03-01:
**Problem Description**
The module version 0.0.19 is failing to be compiled on latest Fedora 41 kernels (6.13.* versions):
```shell
make: Entering directory '/usr/src/kernels/6.13.5-200.fc41.x86_64'
make[1]: Entering directory '/var/lib/dkms/LenovoLegionLinux/0.0.19/build'
  CC [M]  legion-laptop.o
legion-laptop.c:6269:10: error: ‘struct platform_driver’ has no member named ‘remove_new’; did you mean ‘remove’?
 6269 |         .remove_new = legion_remove,
      |          ^~~~~~~~~~
      |          remove
legion-laptop.c:1817:16: warning: ‘ecram_memoryio_write’ defined but not used [-Wunused-function]
 1817 | static ssize_t ecram_memoryio_write(const struct ecram_memoryio *ec_memoryio,
      |                ^~~~~~~~~~~~~~~~~~~~
make[3]: *** [/usr/src/kernels/6.13.5-200.fc41.x86_64/scripts/Makefile.build:196: legion-laptop.o] Error 1
make[2]: *** [/usr/src/kernels/6.13.5-200.fc41.x86_64/Makefile:2024: .] Error 2
make[1]: *** [/usr/src/kernels/6.13.5-200.fc41.x86_64/Makefile:263: __sub-make] Error 2
make[1]: Leaving directory '/var/lib/dkms/LenovoLegionLinux/0.0.19/build'
make: *** [Makefile:263: __sub-make] Error 2
make: Leaving directory '/usr/src/kernels/6.13.5-200.fc41.x86_64'
```

**Model and Debug Info**

Installation type: DKMS
Distribution: Fedora 41
Model name: Lenovo Legion 7 16IAX7
CPU model: Intel Core i7 12800 HX
GPU model: Nvidia RTX 3070
Keyboard backlight: RGB
Light in lid or logo: yes
Light at IO-Ports at back: yes

Output of `sudo dmidecode -t system`. Please remove Serial Number and UUID for privacy:
```text
# dmidecode 3.6
Getting SMBIOS data from sysfs.
SMBIOS 3.4 present.

Handle 0x0001, DMI type 1, 27 bytes
System Information
	Manufacturer: LENOVO
	Product Name: 82TD
	Version: Legion 7 16IAX7
	Wake-up Type: Power Switch
	SKU Number: LENOVO_MT_82TD_BU_idea_FM_Legion 7 16IAX7
	Family: Legion 7 16IAX7

Handle 0x0024, DMI type 12, 5 bytes
System Configuration Options
	Option 1: ConfigOptions1
	Option 2: ConfigOptions2
	Option 3: ConfigOptions3

Handle 0x0027, DMI type 15, 29 bytes
System Event Log
	Area Length: 0 bytes
	Header Start Offset: 0x0000
	Header Length: 8192 bytes
	Data Start Offset: 0x2000
	Access Method: General-purpose non-volatile data functions
	Access Address: 0x0000
	Status: Valid, Not Full
	Change Token: 0x12345678
	Header Format: OEM-specific
	Supported Log Type Descriptors: 3
	Descriptor 1: POST memory resize
	Data Format 1: None
	Descriptor 2: POST error
	Data Format 2: POST results bitmap
	Descriptor 3: Log area reset/cleared
	Data Format 3: None

Handle 0x0034, DMI type 32, 11 bytes
System Boot Information
	Status: No errors detected
```

Output of `sudo dmidecode -t bios`:
```text
# dmidecode 3.6
Getting SMBIOS data from sysfs.
SMBIOS 3.4 present.

Handle 0x0000, DMI type 0, 26 bytes
BIOS Information
	Vendor: LENOVO
	Version: K1CN44WW
	Release Date: 08/17/2023
	Address: 0xE0000
	Runtime Size: 128 kB
	ROM Size: 16 MB
	Characteristics:
		PCI is supported
		BIOS is upgradeable
		BIOS shadowing is allowed
		Boot from CD is supported
		Selectable boot is supported
		EDD is supported
		Japanese floppy for NEC 9800 1.2 MB is supported (int 13h)
		Japanese floppy for Toshiba 1.2 MB is supported (int 13h)
		5.25"/360 kB floppy services are supported (int 13h)
		5.25"/1.2 MB floppy services are supported (int 13h)
		3.5"/720 kB floppy services are supported (int 13h)
		3.5"/2.88 MB floppy services are supported (int 13h)
		8042 keyboard services are supported (int 9h)
		CGA/mono video services are supported (int 10h)
		ACPI is supported
		USB legacy is supported
		BIOS boot specification is supported
		Targeted content distribution is supported
		UEFI is supported
	BIOS Revision: 1.44
	Firmware Revision: 1.36

Handle 0x0025, DMI type 13, 22 bytes
BIOS Language Information
	Language Description Format: Long
	Installable Languages: 8
		en|US|iso8859-1,0
		fr|FR|iso8859-1,0
		zh|TW|unicode,0
		ja|JP|unicode,0
		it|IT|iso8859-1,0
		es|ES|iso8859-1,0
		de|DE|iso8859-1,0
		pt|PT|iso8859-1,0
	Currently Installed Language: en|US|iso8859-1,0
```

Allephh, 2025-03-04:
> See [this Debian bug](https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=1098642).
> 
> TL;DR: fixed in the following commits:
> 
>     * [52bf5d7](https://github.com/johnfanv2/LenovoLegionLinux/commit/52bf5d7efc92b78458f99ba9125b873f0a32ca86)
> 
>     * [ae96a82](https://github.com/johnfanv2/LenovoLegionLinux/commit/ae96a82adb879e729c5727c8769888b47ee1181f)
> 
> 
> You want to have these commits to fix the build in Fedora. What we did in Debian was to package the current git HEAD as these commits are not available in any tagged release yet.

I'm having this error on F41 with kernel 6.13.5: "Failed to start dkms.service - Builds and install new kernel modules through DKMS.".
I don't know how to apply these commits. Is there any forecast for releasing the v0.0.20 tag or have they already been applied in the current release (v0.0.190) and do i need to reinstall?
Thank you in advance.

MrDuartePT, 2025-04-19:
For fedora user the dkms package is now update.

The python-lenovolegionlinux failed to build because of legiond, but it not a huge deal since the python package from v0.0.19 and v0.0.20 is basically the same.

closed 2025-04-19 by MrDuartePT