johnfanv2/LenovoLegionLinux #328: [BUG]

Cat-orii, 2025-06-09:
**Problem Description**
<Please describe your problem here>

Following the README, after entering the `sudo make reloadmodule` command, I get these errors:

```
[Lu iun  9 22:50:07 2025] i915 0000:00:02.0: [drm] *ERROR* Atomic update failure on pipe A (start=208748 end=208749) time 561 us, min 1065, max 1079, scanline start 1006, end 1085
[Lu iun  9 22:56:46 2025] i915 0000:00:02.0: [drm] *ERROR* Atomic update failure on pipe A (start=256619 end=256620) time 574 us, min 1065, max 1079, scanline start 1035, end 1116
...
[Lu iun  9 23:29:36 2025] legion_laptop: Loading legion_laptop
[Lu iun  9 23:29:36 2025] legion PNP0C09:00: legion_laptop platform driver probing
[Lu iun  9 23:29:36 2025] legion PNP0C09:00: Read identifying information: DMI_SYS_VENDOR: LENOVO; DMI_PRODUCT_NAME: 82NL; DMI_BIOS_VERSION:G8CN25WW
[Lu iun  9 23:29:36 2025] legion_laptop: Read identifying information: DMI_SYS_VENDOR: LENOVO; DMI_PRODUCT_NAME: 82NL; DMI_BIOS_VERSION:G8CN25WW
[Lu iun  9 23:29:36 2025] legion PNP0C09:00: is_denied: 0; is_allowed: 1; do_load_by_list: 1; do_load: 1
[Lu iun  9 23:29:36 2025] legion PNP0C09:00: Using configuration for system: G8CN
[Lu iun  9 23:29:36 2025] legion PNP0C09:00: ACPI CFG: 2081289494
[Lu iun  9 23:29:36 2025] legion_laptop: Creating RAM access to embedded controller
[Lu iun  9 23:29:36 2025] legion_laptop: Successfully mapped embedded controller: 0xfe0b0400 (in RAM)/0x0 (in EC) to virtual 0x00000000a62af6d3
[Lu iun  9 23:29:36 2025] legion PNP0C09:00: Read embedded controller ID 0x8227
[Lu iun  9 23:29:36 2025] legion PNP0C09:00: Expected EC chip id 0x5507 but read 0x8227
[Lu iun  9 23:29:36 2025] legion_laptop: Unloading legion ecram
[Lu iun  9 23:29:36 2025] legion_laptop: Unloading legion ecram done
[Lu iun  9 23:29:36 2025] legion_laptop: Unmapping embedded controller memory at 0xfe0b0400 (in RAM)/0x0 (in EC) at virtual 0x00000000a62af6d3
[Lu iun  9 23:29:36 2025] legion_laptop: Unloading legion shared
[Lu iun  9 23:29:36 2025] legion_laptop: Unloading legion shared done
[Lu iun  9 23:29:36 2025] legion PNP0C09:00: legion_laptop not loaded for this device
[Lu iun  9 23:29:36 2025] legion PNP0C09:00: probe with driver legion failed with error -12
```

**Model and Debug Info**

Installation type: via github
Distribution: ManjaroLinux 25.0.3 "Zetar"
Model name: Lenovo Legion 5 15IMH6
CPU model: Intel® Core™ i5-10500H
GPU model: NVIDIA GeForce RTX 3050 Laptop GPU
Keyboard backlight: single color with off/medium/bright)
Light in lid or logo: no
Light at IO-Ports at back: yes (on battery port)

Output of `sudo dmidecode -t system`:
```
# dmidecode 3.6
Getting SMBIOS data from sysfs.
SMBIOS 3.2.0 present.

Handle 0x0001, DMI type 1, 27 bytes
System Information
        Manufacturer: LENOVO
        Product Name: 82NL
        Version: Legion 5 15IMH6
        Serial Number: redacted
        UUID: redacted
        Wake-up Type: Power Switch
        SKU Number: LENOVO_MT_82NL_BU_idea_FM_Legion 5 15IMH6
        Family: Legion 5 15IMH6

Handle 0x001F, DMI type 12, 5 bytes
System Configuration Options
        Option 1: ConfigOptions1
        Option 2: ConfigOptions2
        Option 3: ConfigOptions3

Handle 0x0023, DMI type 15, 29 bytes
System Event Log
        Area Length: 0 bytes
        Header Start Offset: 0x0000
        Header Length: 8192 bytes
        Data Start Offset: 0x2000
        Access Method: General-purpose non-volatile data functions
        Access Address: 0x0000
        Status: Valid, Not Full
        Change Token: 0x12345678
        Header Format: OEM-specific
        Supported Log Type Descriptors: 3
        Descriptor 1: POST memory resize
        Data Format 1: None
        Descriptor 2: POST error
        Data Format 2: POST results bitmap
        Descriptor 3: Log area reset/cleared
        Data Format 3: None

Handle 0x0030, DMI type 32, 11 bytes
System Boot Information
        Status: No errors detected
```

Output of `sudo dmidecode -t bios`:
```
# dmidecode 3.6
Getting SMBIOS data from sysfs.
SMBIOS 3.2.0 present.

Handle 0x0000, DMI type 0, 26 bytes
BIOS Information
        Vendor: LENOVO
        Version: G8CN25WW
        Release Date: 09/20/2023
        Address: 0xE0000
        Runtime Size: 128 kB
        ROM Size: 11 MB
        Characteristics:
                PCI is supported
                BIOS is upgradeable
                BIOS shadowing is allowed
                Boot from CD is supported
                Selectable boot is supported
                EDD is supported
                Japanese floppy for NEC 9800 1.2 MB is supported (int 13h)
                Japanese floppy for Toshiba 1.2 MB is supported (int 13h)
                5.25"/360 kB floppy services are supported (int 13h)
                5.25"/1.2 MB floppy services are supported (int 13h)
                3.5"/720 kB floppy services are supported (int 13h)
                3.5"/2.88 MB floppy services are supported (int 13h)
                8042 keyboard services are supported (int 9h)
                CGA/mono video services are supported (int 10h)
                ACPI is supported
                USB legacy is supported
                BIOS boot specification is supported
                Targeted content distribution is supported
                UEFI is supported
        BIOS Revision: 1.25
        Firmware Revision: 1.25

Handle 0x0020, DMI type 13, 22 bytes
BIOS Language Information
        Language Description Format: Long
        Installable Languages: 8
                en|US|iso8859-1,0
                fr|FR|iso8859-1,0
                zh|TW|unicode,0
                ja|JP|unicode,0
                it|IT|iso8859-1,0
                es|ES|iso8859-1,0
                de|DE|iso8859-1,0
                pt|PT|iso8859-1,0
        Currently Installed Language: en|US|iso8859-1,0
```

Output of `sudo cat /sys/kernel/debug/legion/fancurve`:
```
cat: /sys/kernel/debug/legion/fancurve: No such file or directory
```

sree-vignesh, 2025-07-14:
Hi! Did you have any luck getting the fan control to work on this laptop, I too have the same exact model.

Cat-orii, 2025-07-14:
> Hi! Did you have any luck getting the fan control to work on this laptop, I too have the same exact model.

Hello there.
Nope. Basically nothing works as far as I remember.
Glad that there's also someone trying to help the project 😺

Cat-orii, 2025-08-17:
### **UPDATE**:
Now I've switched to Fedora 42 with KDE and Secure Boot enabled. I was able to sign the kernel module so that LLL will work with Secure Boot.
Now, I have the same problem as mentioned above... BUT, upon further inspection I was able to understand what the problem is. By inspecting the output of `sudo dmesg -T` it seems that the module tried to talk to the embedded controller (EC) and read an EC chip id of 0x8227, but the driver expected 0x5507 for the mapped configuration (system G8CN). Because of that mismatch the driver unloads and returns -12 error from probe. In other words — the driver refuses to load for my device/EC revision. The relevant lines in my logs:
```Read embedded controller ID 0x8227
Expected EC chip id 0x5507 but read 0x8227
legion_laptop not loaded for this device
probe with driver legion failed with error -12
```
I will also attach a file with the output of `sudo dmesg -T | tail -n 200 > ~/dmesg-tail.txt`

[dmesg-tail.txt](https://github.com/user-attachments/files/21821829/dmesg-tail.txt)
 
The output of `umane -a` : 
```Linux Maxi 6.15.9-201.fc42.x86_64 #1 SMP PREEMPT_DYNAMIC Sat Aug 2 11:37:34 UTC 2025 x86_64 GNU/Linux```

And also with this command `sudo make reloadmodule | grep -i error -n > reloadmodule_output.txt` I was able to get every instance of Error that occurs during the making of the module. I will attach the file here.

[reloadmodule_output.txt](https://github.com/user-attachments/files/21821830/reloadmodule_output.txt)

The outputs of `sudo dmidecode -t system` and `sudo dmidecode -t bios` are the same.

Also, with the command `modinfo ~/LenovoLegionLinux/kernel_module/legion-laptop.ko | egrep 'name|vermagic|signer|parm' || true` I was able to identify the **ec_readonly** and **force** parameters:
```filename:       /home/gmaxim/LenovoLegionLinux/kernel_module/legion-laptop.ko
name:           legion_laptop
vermagic:       6.15.9-201.fc42.x86_64 SMP preempt mod_unload 
signer:         Maxi_1755367411_17e5ad1f
parm:           force:Force loading this module even if model or BIOS does not match. (bool)
parm:           ec_readonly:Only read from embedded controller but do not write or change settings. (bool)
parm:           enable_platformprofile:Enable the platform profile sysfs API to read and write the power mode. (bool)
```

So I tried to safely use the **ec_readonly** parameter with `sudo rmmod legion_laptop 2>/dev/null || true` and `sudo insmod /home/gmaxim/LenovoLegionLinux/kernel_module/legion-laptop.ko ec_readonly=1` and then watch the ouptut of dmseg with `sudo dmesg -T | tail -n 80` :
```[Sun Aug 17 12:55:58 2025] Bluetooth: hci0: Found Intel DDC parameters: intel/ibt-19-0-4.ddc
[Sun Aug 17 12:55:58 2025] Bluetooth: hci0: Applying Intel DDC parameters completed
[Sun Aug 17 12:55:58 2025] Bluetooth: hci0: Firmware revision 0.4 build 193 week 33 2024
[Sun Aug 17 12:55:58 2025] Bluetooth: hci0: HCI LE Coded PHY feature bit is set, but its usage is not supported.
[Sun Aug 17 12:55:58 2025] Bluetooth: MGMT ver 1.23
[Sun Aug 17 12:55:58 2025] Generic FE-GE Realtek PHY r8169-0-700:00: attached PHY driver (mii_bus:phy_addr=r8169-0-700:00, irq=MAC)
[Sun Aug 17 12:55:58 2025] r8169 0000:07:00.0 enp7s0: Link is Down
[Sun Aug 17 12:55:58 2025] [drm] Initialized nvidia-drm 0.0.0 for 0000:01:00.0 on minor 0
[Sun Aug 17 12:55:58 2025] nvidia 0000:01:00.0: [drm] Cannot find any crtc or sizes
[Sun Aug 17 12:56:01 2025] r8169 0000:07:00.0 enp7s0: Link is Up - 1Gbps/Full - flow control off
[Sun Aug 17 12:56:01 2025] r8169 0000:07:00.0 enp7s0: Link is Down
[Sun Aug 17 12:56:05 2025] r8169 0000:07:00.0 enp7s0: Link is Up - 1Gbps/Full - flow control off
[Sun Aug 17 12:56:08 2025] Bluetooth: RFCOMM TTY layer initialized
[Sun Aug 17 12:56:08 2025] Bluetooth: RFCOMM socket layer initialized
[Sun Aug 17 12:56:08 2025] Bluetooth: RFCOMM ver 1.11
[Sun Aug 17 12:56:10 2025] Lockdown: systemd-logind: hibernation is restricted; see man kernel_lockdown.7
[Sun Aug 17 12:56:10 2025] Lockdown: systemd-logind: hibernation is restricted; see man kernel_lockdown.7
[Sun Aug 17 12:56:10 2025] Lockdown: systemd-logind: hibernation is restricted; see man kernel_lockdown.7
[Sun Aug 17 12:56:13 2025] systemd-journald[905]: Time jumped backwards, rotating.
[Sun Aug 17 12:56:21 2025] Lockdown: systemd-logind: hibernation is restricted; see man kernel_lockdown.7
[Sun Aug 17 12:56:21 2025] Lockdown: systemd-logind: hibernation is restricted; see man kernel_lockdown.7
[Sun Aug 17 12:56:21 2025] Lockdown: systemd-logind: hibernation is restricted; see man kernel_lockdown.7
[Sun Aug 17 12:56:21 2025] Lockdown: systemd-logind: hibernation is restricted; see man kernel_lockdown.7
[Sun Aug 17 12:56:21 2025] Lockdown: systemd-logind: hibernation is restricted; see man kernel_lockdown.7
[Sun Aug 17 12:56:21 2025] Lockdown: systemd-logind: hibernation is restricted; see man kernel_lockdown.7
[Sun Aug 17 12:56:21 2025] Lockdown: systemd-logind: hibernation is restricted; see man kernel_lockdown.7
[Sun Aug 17 12:56:21 2025] Lockdown: systemd-logind: hibernation is restricted; see man kernel_lockdown.7
[Sun Aug 17 12:56:21 2025] Lockdown: systemd-logind: hibernation is restricted; see man kernel_lockdown.7
[Sun Aug 17 12:56:21 2025] Lockdown: systemd-logind: hibernation is restricted; see man kernel_lockdown.7
[Sun Aug 17 13:11:45 2025] NVRM: nvAssertFailedNoLog: Assertion failed: 0 @ osapi.c:1939
[Sun Aug 17 13:11:50 2025] NVRM: nvAssertFailedNoLog: Assertion failed: 0 @ osapi.c:1939
[... cut ...]
[Sun Aug 17 14:01:44 2025] legion_laptop: Unmapping embedded controller memory at 0xfe0b0400 (in RAM)/0x0 (in EC) at virtual 0x00000000fbcb963a
[Sun Aug 17 14:01:44 2025] legion_laptop: Unloading legion shared
[Sun Aug 17 14:01:44 2025] legion_laptop: Unloading legion shared done
[Sun Aug 17 14:01:44 2025] legion PNP0C09:00: legion_laptop not loaded for this device
[Sun Aug 17 14:01:44 2025] legion PNP0C09:00: probe with driver legion failed with error -12
```
And the driver didn't load either as you can see...

Further instructions will be welcomed 🫡

axeonee, 2025-10-25:
@Cat-orii any more luck since then? Same problem, same model. I see you've done a fair bit more digging than me, I'd be happy to help with testing or anything but I'm too new to all of this to be able to help more

closed 2026-09-11 by github-actions[bot]