johnfanv2/LenovoLegionLinux #334: Support for Legion 5 16IRX9

RedstoneTrail, 2025-06-15:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Model and Debug Info**

Installation type: DKMS from git tree
Distribution: Arch (but that shouldn't affect this I don't think)
Model name: Legion 5 16IRX9
CPU model: Intel i9-14900HX
GPU model: NVIDIA RTX 4060
Keyboard backlight: White (on/low/off)
Light in lid or logo: not in lid (on power button if that counts for logo)
Light at IO-Ports at back: no

Output of `sudo dmidecode -t system`. Please remove Serial Number and UUID for privacy:
```text
# dmidecode 3.6
Getting SMBIOS data from sysfs.
SMBIOS 3.4 present.

Handle 0x0001, DMI type 1, 27 bytes
System Information
	Manufacturer: LENOVO
	Product Name: 83DG
	Version: Legion 5 16IRX9
	Serial Number: XXXXXXXX
	UUID: be0a1f4c-e380-11ef-a4f8-40c2baa6ed26
	Wake-up Type: Power Switch
	SKU Number: LENOVO_MT_83DG_BU_idea_FM_Legion 5 16IRX9
	Family: Legion 5 16IRX9

Handle 0x0024, DMI type 12, 5 bytes
System Configuration Options
	Option 1: ConfigOptions1
	Option 2: ConfigOptions2
	Option 3: ConfigOptions3

Handle 0x0027, DMI type 15, 29 bytes
System Event Log
	Area Length: 0 bytes
	Header Start Offset: 0x0000
	Header Length: 8192 bytes
	Data Start Offset: 0x2000
	Access Method: General-purpose non-volatile data functions
	Access Address: 0x0000
	Status: Valid, Not Full
	Change Token: 0x12345678
	Header Format: OEM-specific
	Supported Log Type Descriptors: 3
	Descriptor 1: POST memory resize
	Data Format 1: None
	Descriptor 2: POST error
	Data Format 2: POST results bitmap
	Descriptor 3: Log area reset/cleared
	Data Format 3: None

Handle 0x0034, DMI type 32, 11 bytes
System Boot Information
	Status: No errors detected
```

Output of `sudo dmidecode -t bios`:
```text
# dmidecode 3.6
Getting SMBIOS data from sysfs.
SMBIOS 3.4 present.

Handle 0x0000, DMI type 0, 26 bytes
BIOS Information
	Vendor: LENOVO
	Version: NMCN30WW
	Release Date: 12/23/2024
	Address: 0xE0000
	Runtime Size: 128 kB
	ROM Size: 16 MB
	Characteristics:
		PCI is supported
		BIOS is upgradeable
		BIOS shadowing is allowed
		Boot from CD is supported
		Selectable boot is supported
		EDD is supported
		Japanese floppy for NEC 9800 1.2 MB is supported (int 13h)
		Japanese floppy for Toshiba 1.2 MB is supported (int 13h)
		5.25"/360 kB floppy services are supported (int 13h)
		5.25"/1.2 MB floppy services are supported (int 13h)
		3.5"/720 kB floppy services are supported (int 13h)
		3.5"/2.88 MB floppy services are supported (int 13h)
		8042 keyboard services are supported (int 9h)
		CGA/mono video services are supported (int 10h)
		ACPI is supported
		USB legacy is supported
		BIOS boot specification is supported
		Targeted content distribution is supported
		UEFI is supported
	BIOS Revision: 1.30
	Firmware Revision: 1.30

Handle 0x0025, DMI type 13, 22 bytes
BIOS Language Information
	Language Description Format: Long
	Installable Languages: 8
		en|US|iso8859-1,0
		fr|FR|iso8859-1,0
		zh|TW|unicode,0
		ja|JP|unicode,0
		it|IT|iso8859-1,0
		es|ES|iso8859-1,0
		de|DE|iso8859-1,0
		pt|PT|iso8859-1,0
	Currently Installed Language: en|US|iso8859-1,0
```

Output of `sudo cat /sys/kernel/debug/legion/fancurve`:
just the default loaded with make forcereloadmodule
```text
EC Chip ID: 5507
EC Chip Version: 2a4
legion_laptop features: fancurve powermode platformprofile platformprofilenotify minifancurve fancurve_pmw_speed fancurve_rpm_speed
legion_laptop ec_readonly: 0
ACPI CFG error: 0
ACPI CFG: 4245487636
temperature access method: 1
CPU temperature error: 0
CPU temperature: 0
CPU temperature EC error: 0
CPU temperature EC: 0
CPU temperature ACPI error: -5
CPU temperature ACPI: 0
CPU temperature WMI error: 0
CPU temperature WMI: 70
CPU temperature WMI2 error: 0
CPU temperature WMI2: 0
CPU temperature WMI3 error: 0
CPU temperature WMI3: 70
GPU temperature error: 0
GPU temperature: 0
GPU temperature EC error: 0
GPU temperature EC: 0
GPU temperature ACPI error: -5
GPU temperature ACPI: 0
GPU temperature WMI error: 0
GPU temperature WMI: 58
GPU temperature WMI2 error: 0
GPU temperature WMI2: 0
GPU temperature WMI3 error: 0
GPU temperature WMI3: 58
fan speed access method: 1
1 fanspeed error: 0
1 fanspeed: 0
1 fanspeed EC error: 0
1 fanspeed EC: 0
1 fanspeed ACPI error: -5
1 fanspeed ACPI: 0
1 fanspeed WMI error: 0
1 fanspeed WMI: 0
1 fanspeed WMI2 error: 0
1 fanspeed WMI2: 0
1 fanspeed WMI3 error: 0
1 fanspeed WMI3: 2400
2 fanspeed error: 0
2 fanspeed: 0
2 fanspeed EC error: 0
2 fanspeed EC: 0
2 fanspeed ACPI error: 0
2 fanspeed ACPI: 2800
2 fanspeed WMI error: 0
2 fanspeed WMI: 0
2 fanspeed WMI2 error: 0
2 fanspeed WMI2: 0
2 fanspeed WMI3 error: 0
2 fanspeed WMI3: 2800
powermode access method: 3
powermode error: 0
powermode: 1
powermode EC error: 0
powermode EC: 230
powermode ACPI error: -5
powermode ACPI: 0
powermode WMI error: 0
powermode WMI: 1
has custom powermode: 1
ACPI rapidcharge error: 0
ACPI rapidcharge: 0
WMI backlight 2 state: 0
WMI backlight 3 state: 0
WMI light IO port: 1
WMI light Y logo/lid: 1
EC minifancurve feature enabled: 1
EC minifancurve on cool: error
EC lockfancontroller error: 0
EC lockfancontroller: true
fanfullspeed error: 0
fanfullspeed: 0
fanfullspeed EC error: -1
fanfullspeed EC: 0
Max speed for fancurve: 10000
Current fan curve in hardware:
Fan curve current point id: 10
Fan curve points size: 10
u(speed_of_unit)|speed1[u]|speed2[u]|speed1[pwm]|speed2[pwm]|acceleration|deceleration|cpu_min_temp|cpu_max_temp|gpu_min_temp|gpu_max_temp|ic_min_temp|ic_max_temp
3	 65	 0	 165	 165	 0	 0	 0	 0	 0	 0	 0	 255
3	 85	 0	 216	 216	 0	 0	 0	 0	 0	 0	 0	 255
3	 22	 0	 56	 56	 0	 0	 0	 0	 0	 0	 0	 255
3	 80	 0	 204	 204	 0	 0	 0	 0	 0	 0	 0	 255
3	 90	 0	 229	 229	 0	 0	 0	 0	 0	 0	 0	 255
3	 28	 0	 71	 71	 0	 0	 0	 0	 0	 0	 0	 255
3	 82	 0	 209	 209	 0	 0	 0	 0	 0	 0	 0	 255
3	 99	 0	 252	 252	 0	 0	 0	 0	 0	 0	 0	 255
3	 28	 0	 71	 71	 0	 0	 0	 0	 0	 0	 0	 255
3	 94	 0	 239	 239	 0	 0	 0	 0	 0	 0	 0	 255
=====================
Current fan curve in hardware (WMI; might be empty)
Fan curve current point id: 0
Fan curve points size: 0
u(speed_of_unit)|speed1[u]|speed2[u]|speed1[pwm]|speed2[pwm]|acceleration|deceleration|cpu_min_temp|cpu_max_temp|gpu_min_temp|gpu_max_temp|ic_min_temp|ic_max_temp
=====================
```

ArcticDev78, 2025-07-28:
Hi, I own the same laptop. Please add support for this specific model ASAP, currently the laptops fans don't work at all using Nobara Linux (KDE Plasma).
 Thank you.

uairhahs, 2025-07-30:
Take a look at [my response to an identical report](https://github.com/johnfanv2/LenovoLegionLinux/issues/344#issuecomment-3137259729).

The maintainer has been a little less active in recent months, hopefully working on good things.

st0nie, 2026-09-01:
Legion 5 16IRX9 (83DG, BIOS NMCN...) is supported since the NMCN allowlist entry landed (model_nmcn). Closing; reopen with logs if something specific does not work.

closed 2026-09-01 by st0nie