johnfanv2/LenovoLegionLinux #337: [BUG] Support for Lenovo Legion Pro 5i Gen 10 (16IAX10H)

tclement0922, 2025-07-05:
**Problem Description**
This laptop does not seem to be supported : the test fan curves values does not match those in the README (wrong EC chip ID, ...).

**Model and Debug Info**

Installation type: github
Distribution: Arch Linux
Model name: 16IAX10H
CPU model: Intel Core Ultra 9 275HX
GPU model: NVIDIA RTX 5070 Ti Laptop
Keyboard backlight: white with off/medium/bright
Light in lid or logo: no
Light at IO-Ports at back: no (and ports are on the side)

Output of `sudo dmidecode -t system`. Please remove Serial Number and UUID for privacy:
```text
# dmidecode 3.6
Getting SMBIOS data from sysfs.
SMBIOS 3.6 present.

Handle 0x0001, DMI type 1, 27 bytes
System Information
        Manufacturer: LENOVO
        Product Name: 83LU
        Version: Legion Pro 5 16IAX10H
        Serial Number: REDACTED
        UUID: REDACTED
        Wake-up Type: Power Switch
        SKU Number: LENOVO_MT_83LU_BU_idea_FM_Legion Pro 5 16IAX10H
        Family: Legion Pro 5 16IAX10H

Handle 0x0024, DMI type 12, 5 bytes
System Configuration Options
        Option 1: ConfigOptions1
        Option 2: ConfigOptions2
        Option 3: ConfigOptions3

Handle 0x0027, DMI type 15, 29 bytes
System Event Log
        Area Length: 0 bytes
        Header Start Offset: 0x0000
        Header Length: 8192 bytes
        Data Start Offset: 0x2000
        Access Method: General-purpose non-volatile data functions
        Access Address: 0x0000
        Status: Valid, Not Full
        Change Token: 0x12345678
        Header Format: OEM-specific
        Supported Log Type Descriptors: 3
        Descriptor 1: POST memory resize
        Data Format 1: None
        Descriptor 2: POST error
        Data Format 2: POST results bitmap
        Descriptor 3: Log area reset/cleared
        Data Format 3: None

Handle 0x0034, DMI type 32, 11 bytes
System Boot Information
        Status: No errors detected
```

Output of `sudo dmidecode -t bios`:
```text
# dmidecode 3.6
Getting SMBIOS data from sysfs.
SMBIOS 3.6 present.

Handle 0x0000, DMI type 0, 26 bytes
BIOS Information
        Vendor: LENOVO
        Version: Q6CN26WW
        Release Date: 05/01/2025
        ROM Size: 48 MB
        Characteristics:
                PCI is supported
                BIOS is upgradeable
                BIOS shadowing is allowed
                Boot from CD is supported
                Selectable boot is supported
                EDD is supported
                Japanese floppy for NEC 9800 1.2 MB is supported (int 13h)
                Japanese floppy for Toshiba 1.2 MB is supported (int 13h)
                5.25"/360 kB floppy services are supported (int 13h)
                5.25"/1.2 MB floppy services are supported (int 13h)
                3.5"/720 kB floppy services are supported (int 13h)
                3.5"/2.88 MB floppy services are supported (int 13h)
                8042 keyboard services are supported (int 9h)
                CGA/mono video services are supported (int 10h)
                ACPI is supported
                USB legacy is supported
                BIOS boot specification is supported
                Targeted content distribution is supported
                UEFI is supported
        BIOS Revision: 1.26
        Firmware Revision: 1.26

Handle 0x0025, DMI type 13, 22 bytes
BIOS Language Information
        Language Description Format: Long
        Installable Languages: 8
                en|US|iso8859-1,0
                fr|FR|iso8859-1,0
                zh|TW|unicode,0
                ja|JP|unicode,0
                it|IT|iso8859-1,0
                es|ES|iso8859-1,0
                de|DE|iso8859-1,0
                pt|PT|iso8859-1,0
        Currently Installed Language: en|US|iso8859-1,0
```

Output of `sudo cat /sys/kernel/debug/legion/fancurve`:
```text
EC Chip ID: 5508
EC Chip Version: 2b0
legion_laptop features: fancurve powermode platformprofile platformprofilenotify minifancurve fancurve_pmw_speed fancurve_rpm_speed
legion_laptop ec_readonly: 0
ACPI CFG error: 0
ACPI CFG: 2081289494
temperature access method: 1
CPU temperature error: 0
CPU temperature: 115
CPU temperature EC error: 0
CPU temperature EC: 115
CPU temperature ACPI error: 0
CPU temperature ACPI: 52
CPU temperature WMI error: 0
CPU temperature WMI: 0
CPU temperature WMI2 error: 0
CPU temperature WMI2: 0
CPU temperature WMI3 error: 0
CPU temperature WMI3: 52
GPU temperature error: 0
GPU temperature: 84
GPU temperature EC error: 0
GPU temperature EC: 84
GPU temperature ACPI error: 0
GPU temperature ACPI: 45
GPU temperature WMI error: 0
GPU temperature WMI: 0
GPU temperature WMI2 error: 0
GPU temperature WMI2: 0
GPU temperature WMI3 error: 0
GPU temperature WMI3: 45
fan speed access method: 1
1 fanspeed error: 0
1 fanspeed: 32190
1 fanspeed EC error: 0
1 fanspeed EC: 32190
1 fanspeed ACPI error: 0
1 fanspeed ACPI: 1700
1 fanspeed WMI error: 0
1 fanspeed WMI: 0
1 fanspeed WMI2 error: 0
1 fanspeed WMI2: 0
1 fanspeed WMI3 error: 0
1 fanspeed WMI3: 1700
2 fanspeed error: 0
2 fanspeed: 15457
2 fanspeed EC error: 0
2 fanspeed EC: 15457
2 fanspeed ACPI error: 0
2 fanspeed ACPI: 1700
2 fanspeed WMI error: 0
2 fanspeed WMI: 0
2 fanspeed WMI2 error: 0
2 fanspeed WMI2: 0
2 fanspeed WMI3 error: 0
2 fanspeed WMI3: 1700
powermode access method: 3
powermode error: 0
powermode: 255
powermode EC error: 0
powermode EC: 147
powermode ACPI error: -5
powermode ACPI: 0
powermode WMI error: 0
powermode WMI: 255
has custom powermode: 1
ACPI rapidcharge error: 0
ACPI rapidcharge: 0
WMI backlight 2 state: 0
WMI backlight 3 state: 0
WMI light IO port: 0
WMI light Y logo/lid: 0
EC minifancurve feature enabled: 1
EC minifancurve on cool: error
EC lockfancontroller error: 0
EC lockfancontroller: false
fanfullspeed error: 0
fanfullspeed: 0
fanfullspeed EC error: 0
fanfullspeed EC: 0
Max speed for fancurve: 10000
Current fan curve in hardware:
Fan curve current point id: 0
Fan curve points size: 0
u(speed_of_unit)|speed1[u]|speed2[u]|speed1[pwm]|speed2[pwm]|acceleration|deceleration|cpu_min_temp|cpu_max_temp|gpu_min_temp|gpu_max_temp|ic_min_temp|ic_max_temp
=====================
Current fan curve in hardware (WMI; might be empty)
Fan curve current point id: 0
Fan curve points size: 10
u(speed_of_unit)|speed1[u]|speed2[u]|speed1[pwm]|speed2[pwm]|acceleration|deceleration|cpu_min_temp|cpu_max_temp|gpu_min_temp|gpu_max_temp|ic_min_temp|ic_max_temp
1        1       0       2       2       0       0       0       0       0       0       0       0
1        3       0       7       7       0       0       0       0       0       0       0       0
1        5       0       12      12      0       0       0       0       0       0       0       0
1        6       0       15      15      0       0       0       0       0       0       0       0
1        7       0       17      17      0       0       0       0       0       0       0       0
1        8       0       20      20      0       0       0       0       0       0       0       0
1        9       0       22      22      0       0       0       0       0       0       0       0
1        10      0       25      25      0       0       0       0       0       0       0       0
1        10      0       25      25      0       0       0       0       0       0       0       0
1        10      0       25      25      0       0       0       0       0       0       0       0
```

bubblegumjoy7, 2026-02-16:
Hello, does anybody have any idea on how make it more usable on this Hardware ?

jxn-30, 2026-09-03:
> This issue has been automatically marked as stale because it has had no activity for 90 days. **If this issue is still affecting you, please leave any comment** (for example, "bump") and it will stay open. It will be closed in 7 days if no further activity occurs.

bump :)

st0nie, 2026-09-14:
Thanks for the report. Your machine is in the EC 0x5508 generation tracked in #491: EC-register reads are misaligned there, so the validated path is a WMI-only model config like `model_q7cn` (added for the 83F5 in #568 — that PR shows exactly what a complete enablement looks like). We don't add allowlist entries without per-model validation because the EC layouts differ per chassis.

To move this forward, please provide (on the current release, no force=1 needed for the dumps):
1. Decompiled DSDT: `sudo cat /sys/firmware/acpi/tables/DSDT > dsdt.dat && iasl -d dsdt.dat` — attach dsdt.dsl
2. From the DSDT: the EC RAM base (ERAX) and the `\_SB_.*.EC0_` device path, and which methods `WMAB` (fan table) implements
3. `sudo dmesg | grep -i legion` with force=1
4. Which WMI sensors are correct (`sensors`, `legion_hwmon` vs `legion-isa`)

With that we can prepare a per-model config for you to test.

tclement0922, 2026-09-19:
Here is the decompiled DSDT : [dsdt.dsl](https://github.com/user-attachments/files/32418209/dsdt.dsl.txt)

And here is the dmesg log:
```
[  799.211195] legion_laptop: Loading legion_laptop
[  799.211278] legion legion: legion_laptop platform driver probing
[  799.211279] legion legion: Read identifying information: DMI_SYS_VENDOR: LENOVO; DMI_PRODUCT_NAME: 83LU; DMI_BIOS_VERSION:Q6CN26WW
[  799.211280] legion_laptop: Read identifying information: DMI_SYS_VENDOR: LENOVO; DMI_PRODUCT_NAME: 83LU; DMI_BIOS_VERSION:Q6CN26WW
[  799.211284] legion legion: is_denied: 0; is_allowed: 0; do_load_by_list: 0; do_load: 1
[  799.211285] legion legion: legion_laptop is forced to load.
[  799.211285] legion legion: legion_laptop is forced to load and would otherwise not be loaded
[  799.211286] legion legion: Using configuration for system: GKCN
[  799.217991] legion legion: Using ACPI device PNP0C09:00 for EC methods
[  799.217993] legion legion: Skipping ACPI _STA check
[  799.217993] legion_laptop: Creating RAM access to embedded controller
[  799.218018] legion_laptop: Successfully mapped embedded controller: 0xfe00d400 (in RAM)/0x0 (in EC) to virtual 0x00000000360cfc1e
[  799.218306] legion legion: Read embedded controller ID 0x5508
[  799.218307] legion legion: Skipped checking embedded controller id
[  799.218307] legion legion: Creating debugfs interface
[  799.218313] legion_laptop: Creating sysfs interface
[  799.218342] legion_laptop: Creating hwmon interface
[  799.218401] legion_laptop: Creating platform profile support
[  799.218427] legion_laptop: Init WMI driver support
[  799.218448] legion_wmi 10AFC6D9-EA8B-4590-A2E7-1CD3C84BB4B1-19: Register after probing for WMI.
[  799.218460] legion_wmi D320289E-8FEA-41E0-86F9-611D83151B5F-20: Register after probing for WMI.
[  799.218482] legion_laptop: Init keyboard backlight LED driver
[  799.218868] legion legion: Led platform::kbd_backlight renamed to platform::kbd_backlight_1 due to name collision
[  799.219225] legion_laptop: Init Y-Logo LED driver
[  799.219785] legion_laptop: Init IO-Port LED driver
[  799.220342] legion legion: legion_laptop loaded for this device
```
As for the `sensors`, `legion_wmi_other` reports correct values (2400 RPM) while `legion_hwmon-isa` reports impossible values (15450 RPM).

tramp-tm, 2026-09-19:
Here is the requested data from my laptop to help implement the WMI-only configuration for this Lenovo generation:

### 1. Decompiled DSDT
I have attached the full [dsdt.zip](https://github.com/user-attachments/files/32420322/dsdt.zip)  file as a ZIP archive to this comment.

### 2. EC Specs & Device Path
* **EC Device Path:** `\_SB.PC00.LPCB.EC0`
<details>
<summary> Full  ERAX Memory Region</summary>

```asl
            OperationRegion (ERAX, SystemMemory, 0xFE500400, 0xFF)
            Field (ERAX, ByteAcc, Lock, Preserve)
            {
                Offset (0x01), 
                VDAT,   8, 
                VSTA,   8, 
                FA2S,   8, 
                FASF,   8, 
                Offset (0x06), 
                FANS,   8, 
                BUSG,   1, 
                BLEG,   1, 
                BATF,   1, 
                BNSM,   1, 
                BTST,   1, 
                BBAD,   1, 
                AUTO,   1, 
                FCHG,   1, 
                STMM,   8, 
                STMH,   8, 
                EDCC,   1, 
                    ,   1, 
                CDMB,   1, 
                CCSB,   1, 
                BTSM,   1, 
                BTCM,   1, 
                    ,   1, 
                Offset (0x0B), 
                SGST,   1, 
                HDMI,   1, 
                    ,   1, 
                    ,   1, 
                    ,   1, 
                    ,   1, 
                    ,   1, 
                Offset (0x0C), 
                ODPO,   1, 
                EODD,   1, 
                ODPK,   1, 
                CMEX,   1, 
                CMON,   1, 
                SODD,   1, 
                ODFB,   1, 
                EODS,   1, 
                RTMP,   8, 
                SBTS,   8, 
                MPTS,   8, 
                PINF,   3, 
                SUPR,   1, 
                GTMP,   1, 
                QUIT,   1, 
                LS35,   1, 
                Offset (0x11), 
                RMBT,   1, 
                RSBT,   1, 
                    ,   2, 
                Offset (0x12), 
                FUSL,   8, 
                FUSH,   8, 
[... 360 lines cut ...]
[1437077.213172] legion_laptop: Init Y-Logo LED driver
[1437077.213892] legion_laptop: Init IO-Port LED driver
[1437077.213937] legion legion: legion_laptop loaded for this device
```

### 4. sensors 
```
tramp@lgn5p-tux-os:~/git/LenovoLegionLinux/kernel_module$ sensors | awk -v RS="" '/^lenovo/'
lenovo_wmi_other-virtual-0
Adapter: Virtual device
fan1:        2400 RPM  (min = 1700 RPM, max = 5300 RPM, div = 100)
fan2:        2100 RPM  (min = 1700 RPM, max = 5300 RPM, div = 100)
tramp@lgn5p-tux-os:~/git/LenovoLegionLinux/kernel_module$ sensors | awk -v RS="" '/^legion/'
legion_hwmon-isa-0000
Adapter: ISA adapter
Fan 1:           18045 RPM  (max = 10000 RPM)
Fan 2:           18023 RPM  (max = 10000 RPM)
CPU Temperature:  +80.0°C  
GPU Temperature:  +87.0°C  
IC Temperature:   +55.0°C
```

### System Info:
* **Laptop Model:** Legion Pro 5 16IAX10H
* **BIOS Version:** Q6CN26WW

st0nie, 2026-09-20:
Support for the Legion Pro 5 16IAX10H (83LU, BIOS Q6CN) has landed in main (commit ad23dd4), thanks to the DSDT and runtime data @tclement0922 and @tramp-tm provided above.

What was added:

- New per-model config `model_q6cn_lu` and a product-qualified DMI allowlist entry (`Q6CN 83LU`: LENOVO + product 83LU + BIOS prefix Q6CN). The Q6CN BIOS is shared with the Legion 5 16IAX10 (83NX, `model_q6cn`), but the 83LU is the Legion Pro 5 chassis, and its DSDT matches the `model_q7cn` (Legion Pro 7 16IAX10H, 83F5) WMI layout.
- Everything fan/thermal goes through WMI, since raw EC reads are misaligned on the 0x5508 generation (#491): fan RPM / CPU+GPU temps / fan full speed via Other Method `WMAE`, power mode via GameZone `WMAA` 0x2C/0x2D, fan table via Fan Method `WMAB` 5/6. The table holds fan LEVELS 1..10 (one table for all fans), so the level-index handling applies and `fan1_level_rpm_table`/`fan2_level_rpm_table` are exposed from `LENOVO_FAN_TABLE_DATA`.
- White keyboard backlight (off/medium/bright) via the KBBACKLIGHT WMI methods; no Y-logo/IO-port lights exist on this chassis, so those attributes are skipped. Rapid charge / battery conservation via `VPC0.GBMD`/`SBMC`.
- `fan_fullspeed` is exposed but writes require custom power mode (same gating as Q7CN/RLCN, where a full-speed write through the wrong path wedged the fans at max until reboot). Power-limit/OC attributes stay hidden except `cpu_temperature_limit`, `cpu_l1_tau` and `gpu_power_target_offset`.

To verify, build the module from current main, `sudo make reloadmodule` (no `force=1` needed anymore), and check:

```
sudo dmesg | grep -i legion    # should match Q6CN 83LU, EC id 0x5508
sensors                        # legion_wmi_other plausible RPM
sudo cat /sys/kernel/debug/legion/fancurve   # u = 5, speed1 in 1..10
```

Details: [83LU notes](https://github.com/johnfanv2/LenovoLegionLinux/blob/main/doc/FEATURES_AND_TESTING.md#legion-pro-5-16iax10h-83lu-q6cn). If anything reads wrong on real hardware, please reopen with the `fancurve` debug dump and `dmesg`.

closed 2026-09-20 by st0nie