johnfanv2/LenovoLegionLinux #344: Allowlist Request :  Legion 5 16IRX9 (83DG)

boubekris, 2025-07-28:
Hello, I tried to install it however it seems to not be in the allowed list. Here are the info:

CPU Model: Intel(R) Core(TM) i9-14900HX
GPU Model: RTX 4070

 output of `sudo dmidecode -t system`:
`# dmidecode 3.3
Getting SMBIOS data from sysfs.
SMBIOS 3.4 present.

Handle 0x0001, DMI type 1, 27 bytes
System Information
	Manufacturer: LENOVO
	Product Name: 83DG
	Version: Legion 5 16IRX9
	Serial Number: MP2PGQKT
	UUID: 256ea6ac-5090-11ef-a4f7-40c2ba65a5ee
	Wake-up Type: Power Switch
	SKU Number: LENOVO_MT_83DG_BU_idea_FM_Legion 5 16IRX9
	Family: Legion 5 16IRX9

Handle 0x0024, DMI type 12, 5 bytes
System Configuration Options
	Option 1: ConfigOptions1
	Option 2: ConfigOptions2
	Option 3: ConfigOptions3

Handle 0x0027, DMI type 15, 29 bytes
System Event Log
	Area Length: 0 bytes
	Header Start Offset: 0x0000
	Header Length: 8192 bytes
	Data Start Offset: 0x2000
	Access Method: General-purpose non-volatile data functions
	Access Address: 0x0000
	Status: Valid, Not Full
	Change Token: 0x12345678
	Header Format: OEM-specific
	Supported Log Type Descriptors: 3
	Descriptor 1: POST memory resize
	Data Format 1: None
	Descriptor 2: POST error
	Data Format 2: POST results bitmap
	Descriptor 3: Log area reset/cleared
	Data Format 3: None

Handle 0x0034, DMI type 32, 11 bytes
System Boot Information
	Status: No errors detected
`

output of `sudo dmidecode -t bios`:

`# dmidecode 3.3
Getting SMBIOS data from sysfs.
SMBIOS 3.4 present.

Handle 0x0000, DMI type 0, 26 bytes
BIOS Information
	Vendor: LENOVO
	Version: NMCN30WW
	Release Date: 12/23/2024
	Address: 0xE0000
	Runtime Size: 128 kB
	ROM Size: 16 MB
	Characteristics:
		PCI is supported
		BIOS is upgradeable
		BIOS shadowing is allowed
		Boot from CD is supported
		Selectable boot is supported
		EDD is supported
		Japanese floppy for NEC 9800 1.2 MB is supported (int 13h)
		Japanese floppy for Toshiba 1.2 MB is supported (int 13h)
		5.25"/360 kB floppy services are supported (int 13h)
		5.25"/1.2 MB floppy services are supported (int 13h)
		3.5"/720 kB floppy services are supported (int 13h)
		3.5"/2.88 MB floppy services are supported (int 13h)
		8042 keyboard services are supported (int 9h)
		CGA/mono video services are supported (int 10h)
		ACPI is supported
		USB legacy is supported
		BIOS boot specification is supported
		Targeted content distribution is supported
		UEFI is supported
	BIOS Revision: 1.30
	Firmware Revision: 1.30

Handle 0x0025, DMI type 13, 22 bytes
BIOS Language Information
	Language Description Format: Long
	Installable Languages: 8
		en|US|iso8859-1,0
		fr|FR|iso8859-1,0
		zh|TW|unicode,0
		ja|JP|unicode,0
		it|IT|iso8859-1,0
		es|ES|iso8859-1,0
		de|DE|iso8859-1,0
		pt|PT|iso8859-1,0
	Currently Installed Language: en|US|iso8859-1,0
`

Output of 'sudo cat /sys/kernel/debug/legion/fancurve':

'(cat: /sys/kernel/debug/legion/fancurve: No such file or directory)'

Thank you very much

uairhahs, 2025-07-30:
Hey, got a Legion 9 16IRX (model 83G0)
Fix that worked for me (keep in mind this is forcing allowlist, use at your discretion):

❯ sudo modprobe -r legion_laptop
❯ sudo modprobe legion_laptop force=1

Then rebuild your boot image and reboot

boubekris, 2025-08-09:
here is the the output of my GPU driver:
`nvidia-smi
Sat Aug  9 10:33:49 2025       
+-----------------------------------------------------------------------------------------+
| NVIDIA-SMI 570.153.02             Driver Version: 570.153.02     CUDA Version: 12.8     |
|-----------------------------------------+------------------------+----------------------+
| GPU  Name                 Persistence-M | Bus-Id          Disp.A | Volatile Uncorr. ECC |
| Fan  Temp   Perf          Pwr:Usage/Cap |           Memory-Usage | GPU-Util  Compute M. |
|                                         |                        |               MIG M. |
|=========================================+========================+======================|
|   0  NVIDIA GeForce RTX 4070 ...    Off |   00000000:01:00.0 Off |                  N/A |
| N/A   44C    P8              2W /   50W |      15MiB /   8188MiB |      0%      Default |
|                                         |                        |                  N/A |
+-----------------------------------------+------------------------+----------------------+
                                                                                         
+-----------------------------------------------------------------------------------------+
| Processes:                                                                              |
|  GPU   GI   CI              PID   Type   Process name                        GPU Memory |
|        ID   ID                                                               Usage      |
|=========================================================================================|
|    0   N/A  N/A            2112      G   /usr/lib/xorg/Xorg                        4MiB |
`

uairhahs, 2025-08-09:
Hmm, this looks fine, I'm now less convinced that the GPU is causing the issue with ACPI.
Checking your dmidecode logs, I see that there are [other reports around ACPI related issues for your model]( https://bbs.archlinux.org/viewtopic.php?id=296214) and [a similar problem on the same repo](https://github.com/johnfanv2/LenovoLegionLinux/issues/84).

Things that might be worth trying:
```shell
#check your platform_profile
lsmod | grep -i platform_profile | awk '{print $4}'
#set your PLATFORM_PROFILE from kernel
PLATFORM_PROFILE=`!!`
```
wether this says legion_laptop or ideapad_laptop might inform why force modprode didn't work, you could try 
```shell
sudo modprobe $PLATFORM_PROFILE force=1
# or use ec_readonly=0 if you don't want to modify and only monitor using the GUI
```
Finally, it may be worth checking Lenovo for a more recent BIOS update.

If none of these work, you will have to wait for the Maintainer to resolve this issue.